DJ2LS/FreeDATA #373: Frequency control display broken (0.8.0-a.1, Windows)

dk5sm, 2023-03-12:
After starting the TNC, the frequency control section shows like this - either frequency and/or mode not showing up:

[image]
[image]

Setting frequency or mode manually is working fine, but it seems that reading out does not work:
[image]

The behavior was observed under Windows only.

Mashintime, 2023-03-12:
This info comes from the TNC to the GUI.  I'd guess we'd need to figure out whether it's the tnc or hamlib.  Did you observe the issue with previous versions of FD?

Maybe ensure you have the latest hamlib and/or running manually rigctld with -vvvv and other args needed for FD

DJ2LS, 2023-03-12:
If it's a version related issue we need to update docs.

I'm running Hamlib 4.4 or so and it's working fine on Ubuntu. Question is, if we can reproduce this with latest Hamlib on Ubuntu as well.

dk5sm, 2023-03-12:
I do not notice this problem on Ubuntu as well. But it´s reproduceable on mine and another Windows-Box. On both machines hamlib is running in version 4.5.4. Same hamlib version used with the previous FD version 0.7.1 / 0.7.2 / 0.7.3 - but without this issue.
We had a similar gui/hamlib issue before. Maybe an unmatched dependency?

Mashintime, 2023-03-12:
I don't experience this issue with 4.5.4 on Windows.  Are both windows boxes using the same transceiver/interface?

dk5sm, 2023-03-12:
No. I am using an IC-705 and on the other machine an IC-7300 is attached. Both via USB-cable.

dk5sm, 2023-03-12:
[image]

I am running rigctld as a windows service on both machines, but running it on the commandline makes no difference.  However, hamlib´s console is throwing an error in an endless loop - "rig_powerstat is not on = 0". Maybe this is a hint?

DJ2LS, 2023-03-12:
Might be the receiving of power status? Introduced with ALC level. That’s the only thing changed in this version Von meinem iPhone gesendet

DJ2LS, 2023-03-13:
Tested on Windows 11 with rigctld 4.5.4 and "dummy" device ( rigctld -m 1 )

@dk5sm could you do me the favour and checking if the problem also occurs with a dummy device?

dk5sm, 2023-03-13:
Sorry, I can´t reproduce the error using the dummy device.

Mashintime, 2023-03-20:
I was able to reproduce on windows with an ic-705.  I think the timeout is occurring too quickly and rigctld can't return the data quickly enough.

Changing the timeout value on line 162 in rigctld.py seems to fix it, but not sure if there might be a better way.

>>self.data_connection.settimeout(0.25)

frspin, 2023-03-20:
Same problem here with Ubuntu 22.04
I am starting rigctld before starting FD with a shell proc.
In Ubuntu 22.04 rigctld is 4.3.1
When FD start there is no info in frequency and in mode fields.
Only after typing a frequency frequency field is populated.

Franco Spinelli
IW2DHW

Mashintime, 2023-03-20:
Please checkout qm-373 branch, it should fix this issue.

https://github.com/DJ2LS/FreeDATA/tree/qm-373

DJ2LS, 2023-03-23:
Should be fixed, @dk5sm , if so, please close this issue

closed 2023-03-24 by dk5sm