DJ2LS/FreeDATA #374: HMAC message signing

DJ2LS, 2023-03-12:
We should sign messages by a shared secret cipher. 

I was thinking about [https://en.wikipedia.org/wiki/HMAC](https://en.wikipedia.org/wiki/HMAC)

Using this library [crypto-js](https://www.npmjs.com/package/crypto-js) and maybe a reduced hash length. 

On IRS we are calculating the hash on received data by using its secret key, exchanged on a different path, maybe phone or email. Means we have to save the shared key in user database I think.

Mashintime, 2023-03-12:
We could go one step further and use PKI.  We could allow users to exchange public keys (even provide a way to allow users to upload their public keys somewhere the client could download them).  Then when sending a message we hash the message, encrypt it with our private key.  On receive client could decrypt the hash with our public key, and verify unecncrypted hash.  This would also provide a means of verifying remote delete requests in issue #364

DJ2LS, 2023-03-12:
But how to verify, a user is the right one behind a callsign? We would still need a key which is exchanged on 2nd communication way, I think. We also should keep hash lengths in mind because of message overhead.

DJ2LS, 2023-03-12:
Using HMAC MD5
```
let CryptoJS = require("crypto-js");
let message = "MESSAGE MESSAGE MESSAGE MESSAGE MESSAGE MESSAGE MESSAGE MESSAGE MESSAGE"
let secret = "123456789"
let hash = CryptoJS.HmacMD5(message, secret)
console.log(hash.toString(CryptoJS.enc.Hex).length)
```

results in a 32byte string

DJ2LS, 2023-08-06:
@Mashintime @LA3QMA 

if we are introducing this on TNC level, it might perform better for headless operation. Any suggestions?

DJ2LS, 2023-08-06:
Idea: 
We are already using a crc32, which will be sent with every message. https://github.com/DJ2LS/FreeDATA/blob/0f1ab974443efb7785e673f413ef8cb9db5bb340/tnc/data_handler.py#L1207 

 It will be then checked on IRS.
https://github.com/DJ2LS/FreeDATA/blob/0f1ab974443efb7785e673f413ef8cb9db5bb340/tnc/data_handler.py#L916C1-L916C1

We can use the crc field for sending the signature instead. On IRS we have to implement a 2-Stage check for the signature/checksum. 
1. Step - check if  signature is equal to the excepted one
1.1 If yes, set message as signed
1.2 if not, proceed with step 2.
2. Step - check if signature is equal to expected checksum
2.1 If yes, set message as unsigned
2.2 If not, message has wrong crc.

closed 2023-09-27 by DJ2LS