DJ2LS/FreeDATA #376: Dark Mode issues

g0hww, 2023-03-18:
There are a few issues with the new Dark Mode, but I like it!

1. Chat dialogue - buttons next to messages, i.e. retry, download, delete, are not visible in dark mode.
2. Chat dialogue - received messages are not darkened
3. Main window - heard stations list entries are not darkened

**General notes on UI**
1. I think it would be good to try and preserve or increase contrast between texts and backgrounds. I prefer peak white on mid-gray or darker backgrounds and pitch black on lighter backgrounds. Mid-gray text is not ideal on any background.
2. The font sizes are a little small in the chat dialogue for things like Ping Acks, etc. Also the bottom line of stats on received messages could be inproved with bigger font.

DJ2LS, 2023-03-18:
@g0hww , thanks for the feedback

Re 1-3: This might be a bug with latest alpha of bootstrap. We might wait some time and see if it's fixed then.

About your general UI notes: Is this regarding to dark or light theme? Yes, I agree, the text size is maybe too small and we should increase it. Maybe we need an adjustment/slider/setting for text size in general....

g0hww, 2023-03-18:
@DJ2LS The general notes apply to both default and dark themes.

Mashintime, 2023-03-19:
@g0hww for the small text of ping/ack/message history and etc.  Is zooming in a satisfactory alternative for you?

Ctrl + + zooms in
Ctrl + - zooms out
Ctrl + 0 resets to 100%

Mashintime, 2023-03-19:
@g0hww I'd recommend looking into some of the dark mode like themes, (I like slate theme, binary waterfall).  I think you may find it more suitable.

DJ2LS, 2023-03-31:
@g0hww any changes needed here?

closed 2023-04-20 by DJ2LS