DJ2LS/FreeDATA #384: Message Store-and-forwarding

jpronans, 2023-03-23:
**Is your feature request related to a problem? Please describe.**
The problem. How to leave a message for someone else to collect when conditions are more favourable.

**Describe the solution you'd like**
This comes from usage of JS8Call and Garim and related to the behaviour of the bands. Having a store-and-forward message facility would be useful to allow one to 'leave' a message or file for someone else to collect later. 

Thinking out loud, it might be easier to get [Garim](https://www.whitemesa.net/garim/garim.html) to support the FreeData TNC.

DJ2LS, 2023-03-23:
Possible duplicate with #382

If we have a failed transmission, we want to retransmit if station is reachable again. Another use case might be, a station is offline and we want to send data as soon as it's available again.

We have two possible ways to go:
Implement in TNC

Could be used in a headless environment
Additional buffer and network API command might needed
Implement in GUI

TNC wouldn't be that blown with features which might cause instability
We could use existing resources/databases for sending data

DJ2LS, 2023-03-25:
@Mashintime , this is a feature, Id like to see, soon. After making the existing version stable enough for a beta, we could start working on this. 
I would be a heavy user of this I think

Mashintime, 2023-03-26:
@DJ2LS yes probably best in GUI.  Will think about best/easiest way to implement......

DJ2LS, 2023-03-26:
I had some thoughts with waiting for BCN frames and some additional information added to each message in database, like actual retries for example for limiting total attempts. 
There have been also some ideas in our Signal group regarding to a time based control so not every beacon is triggering a retransmit. 
We could maybe do some simple maths. Using the existing timestamp vs retries.

DJ2LS, 2023-05-09:
Ah, misunderstood this feature. I now implemented "Store and Retry".
Nevertheless, useful as well I think. 

For store and forward, working completely offline, we might need some kind of "routing tables" or access to stations "heard stations". 

@Mashintime  , any idea? We could do this as part of "request-response" for getting a heard stations list. 
Question here is, if A wants a message to C, using B as relay. How does A know, that the message has been transmitted? 🤔 Sounds like some HF mesh networking....

DJ2LS, 2024-09-27:
This is a needed feature for mesh network