DJ2LS/FreeDATA #401: Last Heard time could be improved

g0hww, 2023-04-22:
Currently, the last heard time only shows "time of day" for the last received traffic from a particular station.  If the software is run for multiple days, it does not inform the operator about which day the time refers to.

There may not be enough space to show a date and time.  Maybe it should just shown the elapsed time since the station was heard. Maybe a config option to set a limit so that stations not heard for a long time are removed from the heard list.  Maybe a tooltip could show the exact date and time if the GUI only displays an elapsed time.  Maybe an elapsed time could show only two units, so "59m59s" changes to "1h0m" and "23h59m" changes to "1d0m".

DJ2LS, 2023-04-22:
....and we should add also a frequency

DJ2LS, 2023-04-22:
Introduced.

closed 2023-04-22 by DJ2LS