DJ2LS/FreeDATA #403: Can we store the chart with the file transfers in the chat, please?

g0hww, 2023-04-23:
It would be nice if we could review the charts for completed transfers, in the chat window.
Perhaps a button for each send and receive.
We could either store images, or raw data.
The info could be deleted when the transfer is deleted from the chat window, either individually or by deleting the entire chat history for that station.

Mashintime, 2023-05-11:
It would be nice, yes.  But the time commitment to implement and test I don't think is warranted at this time.

DJ2LS, 2023-09-30:
we could save the information to the database for each message, but only requesting it when opening the chart. With the new gui this might be easier I think. 
We could add an "info button" next to each message, opening a modal then and displaying the chart. On clicking the info button, we are requesting the chart information from the database.

DJ2LS, 2023-10-16:
Will be implemented in new gui. Some more small steps and it's available

DJ2LS, 2023-10-22:
[image]

closed 2023-10-22 by DJ2LS