DJ2LS/FreeDATA #409: Use FlRig for rig control?

chuchel, 2023-05-06:
Fiiiiirst off, I am not a programmer. Just a keyboard jockey and radio afficionado. If I have no place placing a request here, please let me know before you delete this.

The problem is there are too many configuration possibilities, and limited time to play radio ;-) Hamlib is hard to configure. I have never been able to configure it to work with any of my radios.

Flrig works perfectly right out of it's open source box, and all you need is the port number and speed. I can see all the features of my radio, including ALC meter, SWR meter, even all the filters available on an 891. 

Perhaps you could snip a little from the WSJT code? How does WSJT-X and it's clones control a radio?

Mashintime, 2023-05-06:
Requests are welcomed.

WJST uses hamlib and hamlib can talk to flrig.  Search for the 'flrig' radio model (think it's number 6).  Leave everything else to ignore.  If using the default flrig port should work fine, as I do the same with one of my radios.

Today's radios are complex, the options are necessary even if they don't apply to you.

chuchel, 2023-05-06:
That solves that problem! I had no idea that hamlib had support for flrig. It's number 4 in the latest version of hamlib. I was able to set it to flrig, and ignore all options and it worked right off the bat. Thank you!

chuchel, 2023-05-06:
FlRig is already supported via hamlib.

Mashintime, 2023-05-06:
Great to hear!

closed 2023-05-06 by Mashintime