DJ2LS/FreeDATA #413: When an ARQ completes ok, things can get messy.

g0hww, 2023-05-13:
This issue describes a scenario where many things seem to happen at once. The fact that these things all happen in a short space of time might suggest multiple issues that need to be defined and resolved independently.

When an ARQ session completes successfully, several things might happen, causing overlapping transmissions (clobbering) creating a period of confusion during which nothing useful occurs.

The former ISS might beacon immediately after receiving the final ARQ session Ack from the former IRS, which might also beacon after sending the final Ack. These beacons might collide, causing clobbering.

The former ISS might attempt to open a following ARQ session, again acting in the role of ISS.  The former IRS might now want to send traffic back to  the the same peer station, or potentially to a different peer station so it also starts sending ARQ opens.

It seems that carrier detect based collision avoidance does not function properly.  After an ARQ session with EI7IG as ISS and G0HWW as IRS, I observed in-flight ARQ Open frames coming from EI7IG being clobbered by ARQ Open frame transmissions from G0HWW. **These were synchronised, so that this clobbering continued** until both stations had exhausted their attempts to open an ARQ session.

In a variation of this scenario, the two stations that have just completed an ARQ session may both attempt to open new ARQ sessions with each other and their repeated ARQ Open requests may drift apart so that they are not colliding, but neither station is willing to accept the ARQ open from their peer, as they are both trying to open ARQ sessions in the reverse direction. This scenario repeats until one of both of the peer stations exhausts their retransmission limit and give up.

I think that these factors contribute to the issue described here and they should probably all become individual bug reports, but I wanted to document this one so the scenario is fully explored and considered to be resolved when all contributing factors have been addressed.

1. Carrier detect based collision avoidance does not seem to work, causing clobbering.
2. Lack of random backoff causing cyclic clobbering.
3. Former ISS should probably hold off opening a follow up  ARQ session, by assuming that other stations (including the  former IRS) should now have priority in case they want to use the channel.
4. Beacons should probably not occur directly after an ARQ session ends. The beacon interval timer should probably be reset at the end of an ARQ session so that beacons are sent much later.
5. ARQ open requests should be explicitly rejected by a station that is not in an appropriate state to accept a session request, and such rejects should stop the attempt to open a session by the station that receives them.

DJ2LS, 2023-05-13:
1.) I'm thinking about adding an additional delay, explicitly when receiving a codec2 signal, instead of just a general busy state. This should avoid further collision 

2.) Are you talking about the same like in point 5? Two stations trying to connect at the same time and resulting in a failed connection? 

3.) This could be done by "connecting" to a station and then sending data. Both station will be exclusively busy until disconnect. Within this connection session they can exchange priority data. I removed this feature from gui as it's not necessary for just sending a message. Question is now, how to implement this again in GUI. 

4.) Good idea! Let's do this like you suggested.

5.) This will be already done. Problem here is, if both stations connect at the same time. There is already a random counter which try's to avoid this. But it might be improved if we use codec2 signal receiving state  as well for this 🤔

g0hww, 2023-05-13:
> 2.) Are you talking about the same like in point 5? Two stations trying to connect at the same time and resulting in a failed connection?
Point 5 is one example. Though even for something like a beacon or a CQ, I imagine that collision avoidance needs some kind of random delay otherwise when the channel does go quiet, all stations now think that it is a good time to transmit and they all go at the same time. So perhaps a two step collision avoidance mechanism is needed. First step, is there a signal? If no, then start a short random timer and check again for a signal. If still no, then transmit, else go back to step 1.  Maybe this isn't strictly needed whilst in actively in an ARQ session (as IRS or ISS), to keep the speed up, but it would stop other stations colliding with ARQ traffic.

3) Maybe a simple timer could be used (say 30 secs), which must expire before attempting to open another outgoing  ARQ session. This would give other stations a chance to use the frequency, maybe the previous IRS now wants to send a response, or completely different stations want a chance to do something. Also, maybe the same timer could apply to any ARQ traffic that is received, in order to try and avoid interfering with some other ARQ session between other stations, although I suppose that only signalling traffic from the other ARQ session's IRS might be detected, so maybe this thought is not helpful.

1) Is carrier detect based collision avoidance now complicated by the many different modes, or is there a technique to identify if any codec2 mode is currently being transmitted, even if only the expected mode can actually be decoded?

Mashintime, 2023-05-13:
I think this may all be related to slot based busy detection issue, see #414 

If filter bandwidth is low enough TNC would never go into busy making the built in random time outs useless.  Thoughts?

DJ2LS, 2023-05-13:
3) suggestion, let's reset beacon timer after arq transmission, first. I suggest adding this to "arq_cleanup" function if possible.

4) this depends. In case of codec2, we can mainly use signalling frames for this kind of detection, because we are using the modem state as indicator for this. For general detection "is a codec2 frame present" we might ask @drowe67 and if it's possible doing this independent from actual mode. For common busy detection, FFT is used ( signal strength of a given frequency, in relation to average signal strength ). 

- [ ] @Mashintime the busy detection might be broken when using smaller filters because of splitting fft array with static parameters. Maybe we can do a dynamic calculation for each slot by checking against total fft bandwidth and then using percentage based values for each slot or some calculation, based on Hz/Byte -> not sure if this is equal for different filter bandwidths

Mashintime, 2023-05-13:
No the busy detection was not working because the array containing the data was being overwritten when the first slot was checked clearing out the data for the rest of the ranges, causing subsequent slots to never be busy.  I tested this before and after my changes.  When placing signal in each of the different slots with my changes, it was being triggered in the correct slot.

When station tries to send, and busy is not triggered it'll clobber.  If two stations are both sending connect requests with different random timeout values, the station with the highest random value would transmit even if other station was already transmitting because slot is not being flagged as busy.  Can't just rely on the codec2_data_busy in situations where we can't decode/detect.

Mashintime, 2023-05-13:
Triggered in the correct slot regardless of filter bandwidth

DJ2LS, 2023-05-13:
The. We should test your code, @Mashintime 🙂

DJ2LS, 2023-12-25:
this is outdated as the entire protocol changes

closed 2023-12-25 by DJ2LS