DJ2LS/FreeDATA #419: Add the 27Mhz

maxlinux2000, 2023-05-15:
in europe we can legally use 27Mhz in any digital mode.
Please add 11 meters to the frequency list, thanks
MaX

DJ2LS, 2023-05-15:
@maxlinux2000 are you talking about frequency list used in explorer?

maxlinux2000, 2023-05-16:
> @maxlinux2000 are you talking about frequency list used in explorer?

yes :-)

maxlinux2000, 2023-05-16:
Hello, I still can't tell you anything because I don't have the antenna on the roof. I'm renovating the house and the time hasn't come yet to re-install the antenna... but I got it soon :-)
Until recently I have used js8Call in 11Mt and I have had great experiences. I hope your App is much better.
The idea is to use this app for a Prepper group that we have created.

Soon you will have news.
I will pass this app to my friends

DJ2LS, 2023-05-16:
@maxlinux2000 thanks for your reply! Yes, Js8Call and FreeDATA are completely different I think. What kind of radios are you using and how do you connect them to the computer? This is an area I don't have experience at, because I'm only using ham radio devices and the 11Mt device is already really old. Specially this part is really tricky and I'd like to make it as easy as possible. 
Regarding to your prepper group - We are working on a broadcast / groupchat. Maybe this is useful for you, specially if you have users, which are close together.

maxlinux2000, 2023-05-17:
Hello, the CB radios we use are from the modern ones such as the president mckinleyII and even the old Cobra/CRT/Superstar 3600/3900 from the '90s.
For example, I have made several contacts with Australia from Spain using both a modern CB and an old radio from 30 years ago (super joppix 1000).
As interface we use an adapter similar to EasyDigi... which contains inside a serial<->usb adapter and an Audio<->usb interface. All the interfaces that are used for FT8, psk31, sstv... are valid.

Why CB in 27Mhz? because there are many more people who use these free frequencies than radio amateurs. And you are much more likely to be able to contact on this frequency during an emergency than in all others.
Actually we are thinking of using the PMR446 for this too, but that is another matter.

Regards
MaX

closed 2023-05-22 by DJ2LS