DJ2LS/FreeDATA #424: LGPL may not be the best choice of licence.

g0hww, 2023-05-18:
LGPL is typically applied to open source libraries that wish to allow for linking by GPL incompatible applications.
For this project, the regular GPLv3 is probably a better choice.

The good news is that LGPL says 

> "You may opt to apply the terms of the ordinary GNU General Public License instead of this License to a given copy of the Library. To do this, you must alter all the notices that refer to this License, so that they refer to the ordinary GNU General Public License, version 2, instead of to this License." 

so there is no need to ask all contributors.

DJ2LS, 2023-05-22:
@g0hww, @drowe67  Meanwhile I moved to GPL v3, thanks @g0hww to your notification about this. Nevertheless we need to check for compatibility between GPL v2 ( codec2 ) and GPL v3 ( FreeDATA ). 

In my opinion there shouldnt be a problem, because we are using libcodec2 as compiled version without changes, so it acts like a "own program" - or am I wrong?

https://www.gnu.org/licenses/rms-why-gplv3.en.html
```
When we say that GPLv2 and GPLv3 are incompatible, it means there is no legal way to combine code under GPLv2 with code under GPLv3 in a single program. This is because both GPLv2 and GPLv3 are copyleft licenses: each of them says, “If you include code under this license in a larger program, the larger program must be under this license too.” There is no way to make them compatible. We could add a GPLv2-compatibility clause to GPLv3, but it wouldn't do the job, because GPLv2 would need a similar clause.

Fortunately, license incompatibility matters only when you want to link, merge or combine code from two different programs into a single program. There is no problem in having GPLv3-covered and GPLv2-covered programs side by side in an operating system. For instance, the TeX license and the Apache license are incompatible with GPLv2, but that doesn't stop us from running TeX and Apache in the same system with Linux, Bash and GCC. This is because they are all separate programs. Likewise, if Bash and GCC move to GPLv3, while Linux remains under GPLv2, there is no conflict.
```

g0hww, 2023-05-22:
There is no problem with compatibility between codec2's LGPL licence and the GPLv3. The quote you provide specifically relates to the GPLv2 and GPLv3, not LGPL.

See sections on LGPL here:
https://www.gnu.org/licenses/license-list.en.html

DJ2LS, 2023-05-22:
Also found this one:
https://www.gnu.org/licenses/gpl-faq.html#AllCompatibility

Okay, problem fixed, closing this one. Thanks for your help here, @g0hww