DJ2LS/FreeDATA #429: Bug: Failed info session.

jpronans, 2023-05-21:
**Describe the bug**
During a "info request" session. My station appeared to retry sending an old message to a different callsign even though they hadn't been heard in a few hours

**To Reproduce**
Unsure

**Expected behavior**
That the "station info" request session would complete.

**Logfiles**
2023-05-21T20:55:27.971406Z [debug    ] [TNC] send_data_to_socket_queue: jsondata={"freedata":"tnc-message","arq":"transmission","status":"opened","mycallsign":"EI7IG-0","dxcallsign":"G0HWW-0","irs":"True"}
2023-05-21T20:55:27.972106Z [info     ] [TNC] ARQ | DATA | RX | [EI7IG-0]>>|<<[G0HWW-0] bandwidth=wide snr=3.1
2023-05-21T20:55:28.968851Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=3 sync_flag=--ST
2023-05-21T20:55:29.061845Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=3 sync_flag=--ST
2023-05-21T20:55:29.167712Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:29.262220Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:29.367669Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:29.460505Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:29.563989Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:29.666061Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:29.770131Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:29.863677Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:30.068587Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:30.163809Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:30.268043Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:30.363457Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:30.466480Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:30.559447Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:30.664572Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:30.770666Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:30.864369Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:30.967937Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:31.165466Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:31.269429Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:31.363513Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:31.466609Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:31.559701Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:31.664544Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:31.770785Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:31.864833Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:31.967943Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:32.061163Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:32.269253Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:32.363738Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:32.466761Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:32.559854Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:32.664720Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:32.770627Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:32.866565Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=2 sync_flag=--S-
2023-05-21T20:55:33.198084Z [debug    ] [MDM] [demod_audio] modem state mode=dat0-datac1 rx_status=6 sync_flag=-BS-
2023-05-21T20:55:33.199602Z [debug    ] [MDM] [demod_audio] Pushing received data to received_queue nbytes=512
2023-05-21T20:55:33.200537Z [debug    ] [MDM] worker_received: received data!
2023-05-21T20:55:33.201347Z [debug    ] [TNC] process_data:            n_retries_per_burst=0
2023-05-21T20:55:33.202115Z [debug    ] [HLP] check_callsign: Checking: callsign=b'EI7IG-0'
2023-05-21T20:55:33.204738Z [debug    ] [HLP] check_callsign: Checking: callsign=b'EI7IG-0'
2023-05-21T20:55:33.206547Z [debug    ] [HLP] check_sessionid: Checking: check=b'\x01' ownid=b'.'
2023-05-21T20:55:33.207337Z [debug    ] [HLP] check_sessionid: Checking: check=b'.' ownid=b'.'
2023-05-21T20:55:33.208060Z [debug    ] [TNC] RX SNR                   snr=3.1
2023-05-21T20:55:33.208943Z [debug    ] [TNC] temp_burst_buffer        buffer=b'BOF\x90\xf8\t\xe6\x00\x00\x00\r\x01\xfd7zXZ\x00\x00\x04\xe6\xd6\xb4F\x02\x00!\x01\x16\x00\x00\x00t/\xe5\xa3\x01\x00\x0cm\x00;\x01;req\x00;\x01;0\x00\x00\x00\x00\x081\xe9de%.\xaa\x00\x01%\rq\x19\xc4\xb6\x1f\xb6\xf3}\x01\x00\x00\x00\x00\x04YZEOF\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
2023-05-21T20:55:33.210252Z [debug    ] [TNC] ARQ.rx_frame_buffer      buffer=b''
2023-05-21T20:55:33.210959Z [debug    ] [TNC] ARQ | RX | appending data to buffer
2023-05-21T20:55:33.211919Z [debug    ] [TNC] arq_data_received:       bof_position=0 eof_position=84
2023-05-21T20:55:33.212993Z [info     ] [TNC] ARQ | RX | DATA FRAME SUCCESSFULLY RECEIVED bytesperminute=5805 duration=5.240276336669922 nacks=0 total_bytes=13
2023-05-21T20:55:33.226088Z [info     ] [TNC] ARQ | RX | saving data to rx buffer buffer_size=8 maxsize=16
2023-05-21T20:55:33.238443Z [debug    ] [TNC] send_data_to_socket_queue: jsondata={"freedata":"tnc-message","arq":"transmission","status":"received","uuid":"c17e2212-8bc5-4fcc-86c7-d3b7132abec2","perc [... cut ...]
2023-05-21T20:55:34.239517Z [info     ] [TNC] ARQ | RX | SENDING DATA FRAME ACK
2023-05-21T20:55:34.265093Z [debug    ] [TNC] enqueue_frame_for_tx     c2_mode=sig0 data=[bytearray(b'=.\x1f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')] type=FR_ACK
2023-05-21T20:55:34.288602Z [debug    ] [MDM] TRANSMIT                 delay=0 mode=19 payload=14

**Screenshots**
-

**Desktop (please complete the following information):**
TNC on Raspberry Pi,
Desktop on Pop-OS 22.04 Laptop

**Additional context**
Add any other context about the problem here.

DJ2LS, 2023-05-22:
Thanks for the report, @jpronans ! 

We might keep this in mind for checking if it occurs again. I have no idea why this happened.

g0hww, 2023-05-22:
I have to come to describe the scenario in which this happened.

After a day of poor propagation, when John's station eventually appeared with a good SNR, I decided to fetch his profile and go to slump on the sofa.

My request to John's station was delivered OK, and after a brief pause, his TNC attempted to open an ARQ connection, but not to my station! Looking at the TNC console output here, I could tell that his TNC was trying to open a session, but it was considered to be foreign, not addressed to my station. There was no response to those attempts to open an ARQ and eventually his TNC gave up.

I suspected that John's TNC attempted to open an ARQ session with whatever station happened to be selected in his chat window, instead of my station.

This seemed like a repeat of a similar occurrence some time ago,, in which I tried to fetch Simon's profile and observed what appeared to be his TNC immediately attempt to open an ARQ session with some unknown 3rd station.

I hope this info is useful.

DJ2LS, 2023-05-23:
Todo: check if Callsign selection can affect response request

Mashintime, 2023-07-12:
I observed this myself and should be fixed in the most recent alpha.

closed 2023-07-12 by Mashintime