DJ2LS/FreeDATA #478: Universal all for macOS

djbr1, 2023-09-18:
Hello

Please consider compiling and releasing Universal app for macOS 

73
HB9TXB

DJ2LS, 2023-09-30:
Saving some notes:

https://github.com/github/roadmap/issues/528

DJ2LS, 2023-10-23:
I added ["arm64", "x64"] to build steps, maybe the next release contains a working version :thinking: @djbr1

djbr1, 2023-10-24:
Hello
despite changes in electron-builder,  yesterdays release (FreeDATA-Mac-0.11.0-alpha.6-Installer.dmg) is  not universal:
file /Applications/FreeDATA.app/Contents/MacOS/FreeDATA
/Applications/FreeDATA.app/Contents/MacOS/FreeDATA: Mach-O 64-bit executable x86_64

DJ2LS, 2023-10-24:
@djbr1 lets wait for the next release, I'm not sure if its working then, we need several attempts maybe.

DJ2LS, 2023-10-24:
@djbr1 I pushed another release, containing some changes, I expect it is failing for universal, but I need to gather some experience here. I'm not sure how to bundle the app because I'm using some build tools which are not always including support for our use cases. Another option could be, providing new packages for Apple, means we will break compatibility with existing versions, but then we would have an x64 package and an arm64 one

DJ2LS, 2023-10-25:
not sure if the latest merge will fix this. It's not that well documented. But it's another attempt.  On my local machine, I was able creating an universal build, but for some reason this isnt working for the GitHub build action....

DJ2LS, 2023-10-29:
This causes really headache. But we are making slowly progress. A hot fix is pushed for the build process, let's see if it is working for macOS as well. 
While it seems we are making progress in case of electron universal app, we also have to create universal builds for the modem. Problem here: Pyinstaller doesnt support easy building of universal apps yet....

djbr1, 2023-10-29:
Thanks for your effort. Please de-prioritize the issue.  In case someone has Mac M1/M2 and wants to run natively, you may send him binaries directly from your dev environment.

DJ2LS, 2023-10-29:
...and compiling from source on Mac isn't that hard!

DJ2LS, 2024-07-27:
Meanwhile things have changed a bit. We are dropping the GUI application, which caused lots of headache and now we only have to focus on deploying the server, as it contains the GUI now ( via web interface / browser ). Question is now, how will the precompiled bundle work, specially because of MacOS code notarization 🤔

DJ2LS, 2024-09-15:
Meanwhile the application design has changed a bit. It's only needed running freedata in a cli, the gui can be accessed then via browser. 

I'm actually doing some tests with a pip package on my Mac here, as it seems, we can avoid doing code signing here and are not dependent on other packages like brew. Both parts are not that common on MacOS, as they are options for advanced users. 

@deejdeej do you have another idea, how we could use scripts on macOS, without further dependencies? Installing packages like nodejs might ne be that trivial like on Linux. Also the MacOS world doesn't have that much differences like the Linux world, so just providing a precompiled binary for codec2 might be enough ( like we are doing it on Windows )

artstarz, 2024-12-29:
Hi.. brew for "ham radio user" on macbook this no problem.

DJ2LS, 2025-02-27:
short update: we have now installation scripts using brew.

Code signing on Apple platform is still really expensive and a pain to maintain. So question its, if we should keep this feature request open?