DJ2LS/FreeDATA #479: The program is very buggy.

Svetlin19777, 2023-09-30:
The program is very buggy. everything sucks. does not want to connect to a transmission for ic-7600. From version Free DATA-Setup-0.8.1-alpha upwards does not find the sound or usb ports in any way.
I use windows 10 x64.

DJ2LS, 2023-09-30:
Hi @Svetlin19777 , thanks for taking the time and opening an issue! I understand, you're possibly frustrated, but your wording and topic description are not helping that much. Also the software is still an alpha. So errors are common. 

Did you have a look at the wiki, specially here: https://wiki.freedata.app/en/usage/gui-config-and-logging-files ? 

Also please try deleting the config folder and then starting the app again. 

If you need help in any case, free free asking on our discord channel: https://discord.gg/Zgqd7hrTtD

Thanks, Simon

Svetlin19777, 2023-10-02:
Thanks for the email. My description is not good because my English is not
good. I'm sorry. I got to know everyone about the program and the launch.
Including additional radio tuning program. But from the version I
specified, it doesn't want to find the sound cards and usb ports.
Apparently the registry it creates on my computer stays somewhere. I dont
know where. And the settings are transferred automatically when running a
version. This creates a problem. I followed the setup method step by step.
Only once was it able to drive my IC-7600 radio to transmit. It was on some
old version. So there is hope. :)

Svetlin19777, 2023-10-02:
Hello again.
Today I was able to set up the program to connect to my ic-7600 radio. I am
sending photos of the settings showing the version the radio is running. I
note that I had to run the program several times until the test button
turned green. There is a problem in this direction and it needs to be
worked on.
Good luck!

Svetlin19777, 2023-10-02:
noting audio devices for this version

Svetlin19777, 2023-10-02:
interesting, i pressed the stop button and the ports appeared.

Svetlin19777, 2023-10-02:
[image: image.png]

DJ2LS, 2023-10-02:
Hi, @Svetlin19777 no worries about your english skills! I'm actually reworking the GUI, so this needs to be tested again then as soon as the new gui is published. THe old one has several problems. 

FOr performancy reasons, the "daemon" which is gathering the audio device information is only working, while tnc is stopped.

DJ2LS, 2023-10-17:
I'm closing this, as its deprecated because of work on new gui

closed 2023-10-17 by DJ2LS