DJ2LS/FreeDATA #485: Package dependencies on Raspberry Pi for emojis

deejdeej, 2023-10-20:
To get the emojis to appear correctly in the chat sessions on a Raspberry Pi, install the packages:
node-emoji
node-emojis-list
fonts-noto-color-emoji

Debian GNU/Linux 11 (bullseye)

deejdeej, 2023-10-25:
@DJ2LS They are needed in Raspberry Pi OS 11 which most people are likely running today.  Raspberry Pi OS 12 just came out 2 weeks ago and I haven't had a chance to try it yet.

DJ2LS, 2023-10-26:
I'm not sure about the node-emojis-list dependency

DJ2LS, 2023-10-26:
fonts-noto-color-emoji cant be found on desktop systems

DJ2LS, 2023-10-26:
@deejdeej can you test with @electron-fonts/noto-color-emoji ?

deejdeej, 2023-10-26:
I'm running the 64 bit "Raspberry PI OS with desktop" version for both 11 and 12.  An "apt install fonts-noto-color-emoji" works on both.

Can you explain what you mean by "fonts-noto-color-emoji cant be found on desktop systems" ?

DJ2LS, 2024-02-06:
@deejdeej do we have a actual status on this?

deejdeej, 2024-02-06:
"apt install fonts-noto-color-emoji" works on both Raspberry Pi OS 11 and 12, and I have verified this package is needed for the emoji fonts to display correctly in the FreeDATA GUI.

commit c650b3c9 (2024-02-08): adjusted raspberrpi emoji package, closes #485
closed 2024-02-08 by DJ2LS