DJ2LS/FreeDATA #505: Instructions for building on Raspberry Pi OS 12 64 bit (Debian bookworm)

deejdeej, 2023-10-25:
# Instructions to install FreeDATA on a fresh installation of 
# "Raspberry Pi OS with desktop" version 12 (Debian bookworm)

# Permissions:
sudo adduser $USER dialout
logout, then log back in

# Dependencies:

sudo apt install libhamlib-utils libhamlib-dev libhamlib4 fonts-noto-color-emoji
sudo apt install git build-essential cmake npm
sudo apt install nodejs
sudo apt install python3 portaudio19-dev python3-pyaudio python3-pip python3-colorama

# Create virtual Python environment:

cd
python3 -m venv ~/venv
~/venv/bin/pip install --upgrade pip wheel

# Install FreeDATA:

git clone https://github.com/DJ2LS/FreeDATA.git
cd FreeDATA
~/venv/bin/pip install -r requirements.txt
cd gui
npm i
cd ..

# Install codec2:

cd codec2
mkdir build_linux
cd build_linux
cmake ..
make -j4
cd

# To run the software, open two terminal windows.

# In the first terminal, run the modem:
cd ~/FreeDATA/modem
source ~/venv/bin/activate
python3 daemon.py

# In the second terminal, run the GUI:
cd ~/FreeDATA/gui
npm start

DJ2LS, 2023-10-25:
Thanks, @deejdeej 

@kronenpj are you interested in adjusting the wiki?

LA3QMA, 2023-10-26:
I can fix this

LA3QMA, 2023-10-26:
Updated

closed 2023-10-26 by LA3QMA