DJ2LS/FreeDATA #551: Please An Android version of this application

Tfjjjastro, 2023-12-02:
I mean, you haven't included Android and also if you are planing for Android then definitely give Bluetooth spp Or rfcomm support to use these app features and pls try to include opus encoding and decoding with real time voice walkie talke using KISS protocol Or any reliable protocol

DJ2LS, 2023-12-02:
Hi @Tfjjjastro , we are actually rewriting the application so it can act more as server. This results in much easier data handling for third party apps. Means, an app for mobile devices isn't unrealistic. But work didn't start yet and we have to finish the server parts first as they are the core of everything. 

KISS related protocols fit more to the packet radio world, last tests haven't been that promising because the kiss protocols payload doesn't directly that well to the used LDPC codes used for the mode FEC. So this part is a future topic. But I understand the interest in a maximum of interoperability and portability. Maybe we'll find a tradeoff somewhen

DJ2LS, 2024-07-23:
@Tfjjjastro For now the server is running only on desktop operating systems, but with the upcoming release it will be possible, opening the GUI on mobile devices as well

closed 2024-08-03 by DJ2LS