DJ2LS/FreeDATA #646: Add Support for Pulseaudio

iqw1345, 2024-02-11:
When adding an input or output audio device using the JACK Audio Connection Kit on Linux, I get an invalid sample rate error. I am confident that this is not an issue with FreeDATA, rather pipewire or my own system. I get similar errors from fldigi when trying to use JACK as well. However, I can set both audio devices to pipewire[ALSA] and run something like qjackctl to manually connect FreeDATA to the radio, but it's tedious. Ideally, similar to WSJT-X, having pulseaudio devices be listed would be the best solution. 

Attached is WSJT-X on Linux listing pulse devices.
[image]

If any further elaboration or clarification is needed, please feel free to let me know.

Mashintime, 2024-02-11:
@iqw1345 thanks for the info.

@DJ2LS I've lost the reference page I saw this on, but they recommend querying the device with the proper backend before initializing it.  I don't know if maybe we could blindly query for the device by backend, and they try initializing it if that'd help.

DJ2LS, 2024-02-11:
Portaudio already has a pulse device, I'm using for my radio setup. Sure, the entire native pulse support is coming in the next version of portaudio. Not sure how it's looking then like. But basic support for selecting pulse as device and configuring it via pavucontrol is working. 

What confuses me a bit is, that no pulse device is available?

iqw1345, 2024-02-11:
Firstly, I should have started out by saying that I am running pipewire. I do have pulse[ALSA] available, but it usually defaults to my regular speakers.

Attached is FreeDATA listing audio devices:
[image]

Actually, I was able to get the ICOM 7300 [JACK Audio Connection Kit] working now, after closing down all radio applications and running:
systemctl --user restart pipewire

The JACK and Pulse applications seem to be working in harmony now. Sorry about that. Should have tested that first.

Having pulse support would be nice, but seeing as how people are moving to pipewire, which basically integrates everything, I wouldn't mind closing the ticket, but I'll leave it to y'all to decide how to proceed.

DJ2LS, 2024-02-12:
Good it's working, if it's okay for you we can close this after @LA3QMA added this to the knowledge base. I think we should document this command as it solves some errors it seems.

iqw1345, 2024-02-12:
Sounds great to me.

DJ2LS, 2024-02-26:
@LA3QMA did you see this one? We can close it then 🙂

LA3QMA, 2024-02-26:
You can close. I'll add it to trello and update the wiki later

closed 2024-03-03 by DJ2LS