DJ2LS/FreeDATA #649: Beacon interval change results in modem crash

OH2JRB, 2024-02-14:
**Describe the bug**
Modem does not restart when beacon interval is changed using GUI. Requires manual intervention.

**To Reproduce**
Change beacon interval in GUI (in Modem tab).

**Expected behavior**
Clean restart.

**Screenshots**
2024-02-14 10:11:18 [info     ] [RIGCTLD] Connected to rigctld at 127.0.0.1:4532
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( hostApi, parameters, streamDir, &pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1768
2024-02-14 10:11:19 [warning  ] [Audio-Test] Input device error (2f83): e=PortAudioError('Device unavailable', -9985)
2024-02-14 10:11:19 [debug    ] [MDM] [demod_audio] demod loop ended e=PortAudioError('Invalid stream pointer', -9988) mode=DATAC3
2024-02-14 10:11:19 [debug    ] [MDM] [demod_audio] demod loop ended e=PortAudioError('Invalid stream pointer', -9988) mode=DATAC4
2024-02-14 10:11:19 [debug    ] [MDM] [demod_audio] demod loop ended e=PortAudioError('Invalid stream pointer', -9988) mode=DATAC1
2024-02-14 10:11:19 [debug    ] [MDM] [demod_audio] demod loop ended e=PortAudioError('Invalid stream pointer', -9988) mode=DATAC0
2024-02-14 10:11:19 [debug    ] [MDM] [demod_audio] demod loop ended e=None mode=SIGNALLING
2024-02-14 10:11:19 [info     ] tested audio devices           result=[False, True]
2024-02-14 10:11:19 [warning  ] starting modem failed          input_test=False output_test=True
2024-02-14 10:11:19 [debug    ] Event:                         ev={'modem': 'failed'}

**Desktop (please complete the following information):**
Intel NUC running Debian Bookworm 12.5
FreeDATA (branch: develop): GUI 0.13.4-alpha. Modem 0.13.4-alpha.

DJ2LS, 2024-02-19:
I think we should remove the beacon interval setting in future releases

OH2JRB, 2024-02-25:
Instead of removing this feature: please consider doing the beacon interval change without restarting the modem. Secondly, please fire a beacon when this feature is turned on (as it was previously).

DJ2LS, 2024-02-25:
Problem / thought with beacon interval is, that we want to use it for mesh networking in future. Experiments in past have shown, that a too short beacon interval causes a really high channel traffic load. So we should consider increasing the interval to about 5minutes for not flooding the channel too much. 

Problem with sending the beacon after activating is, that it can be abused. For sending a "hey I'm there" I recommend using the CQ/QRV or ping. But I'm open to further feedback here. What's the specific reason why the beacon should transmit immediately? ( now it is transmitting after a specified time )

DJ2LS, 2024-03-03:
@LA3QMA @Mashintime @dk5sm any thoughts regarding to this?

dk5sm, 2024-03-03:
Changing the beacon interval (using both server and GUI 0.13.6 under Windows) does not lead to some kind of crash, at least it does not for me.

As far as the beacon time with comparable software such as JS8 or VarAC is concerned, the minimum time here is 10 or 15 minutes. I would find a minimum of 5 minutes and a default of 10 minutes for FD perfectly acceptable. I wouldn't want to do without the function itself either, because it provides an essential view of the propagation conditions.

With regard to longer beacon intervals, I would also like the beacon to be sent out immediately, as otherwise you would have to wait for the interval time to be able to read the first results in the explorer. In case of doubt, a switch in the settings "send beacon immediately" would possibly help here (compromise?).

DJ2LS, 2024-03-03:
My intention is, not bloating the code with unnecessary code and keeping it simple where possible.

A 10min interval would be fine, but we should hardcode it so we can reduce the amount of settings. I also don't see a reason why a beacon interval needs to be changed. But I might be wrong, so please give me a use case if there's one.

The initial transmission when enabling the beacon might make sense if we are talking about possible future features like a "away status", when changing it, we should send an updated beacon. Thats the only reason making sense zo me for now, 
for all the other cases a CQ can or should be used in my opinion.

jpronans, 2024-03-03:
I'd be slow to hard-code it myself, 10 minutes is not bad at all, just to add confusion, you could also consider an initial interval of 5 on a status change then 10,20,30mins for an unattended station?

DJ2LS, 2024-03-07:
Okay, I'm thinking about some future stuff as well, but I want to close this one, next days. 

- I will incerase the beacon interval to 10minutes for reducing channel load. 
- And a beacon will be triggered as soon as you enable it.

Is everyone fine with this, @OH2JRB @dk5sm @jpronans