DJ2LS/FreeDATA #662: Radiocontrol via HamRadio deluxe

linux150, 2024-02-23:
Would be nice, because i use it for all other digimodes too with portsplitter. Now working with flrig but HRD would be nice.

73
Martin

DJ2LS, 2024-02-25:
Martin, do you have any further information about hrd radio server and the corresponding api?

linux150, 2024-02-25:
Sorry, not really
Js8call is opensource and work with HRD.
Maybe Irad will also use it on VarAc, but not programmed until yet.
73

DJ2LS, 2024-02-26:
I contacted HRD, FYI

DJ2LS, 2024-03-11:
still no response from HRD

DJ2LS, 2024-07-27:
Contacted HRD again

DJ2LS, 2024-07-30:
got a response
result: they lost the api documentation, I should check the WSPR source code. 

To me it seems, it's not making sense, continuing with this as long as there is no interest of HRD.

closed 2024-07-30 by DJ2LS