DJ2LS/FreeDATA #674: modemport fixed or defined by user?

frspin, 2024-03-04:
Is modemport setting in config file for modem used?
It seem that 5000 as port is fixed and value in config file ignored.

Is this correct?
If is fixed we can not have different modem on different rig at the same time.

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2024-03-04:
hi @frspin the modem port should be adjustable, but there are some problems actually occuring with this. We will fix this from time to time.

frspin, 2024-03-04:
Il 04/03/24 14:55, DJ2LS ha scritto:
> hi @frspin <https://github.com/frspin> the modem port should be 
> adjustable, but there are some problems actually occuring with this. We 
> will fix this from time to time.

Ok
I will wait for this.

Please keep me informed on fix

Franco Spinelli
IW2DHW

DJ2LS, 2024-03-04:
please check the message I've sent you via HF ;-) 

the "develop" branch contains latest changes

closed 2024-03-04 by DJ2LS