DJ2LS/FreeDATA #676: Can't find the sound card.

pepefrog1234, 2024-03-07:
**Describe the bug**
Can't find the sound card.

**To Reproduce**
Steps to reproduce the behavior:
1.Install the latest version
2. Go to the modem settings page
3. The sound card does not appear in the sound card drop-down menu.

**Expected behavior**
A clear and concise description of what you expected to happen.
Find the sound card that your computer currently has.

**Logfiles**
https://wiki.freedata.app/en/usage/logging

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**

- OS: Windows
- Version 10

DJ2LS, 2024-03-07:
@pepefrog1234 thanks for the report! Which FreeDATA are you running?

pepefrog1234, 2024-03-07:
0.14.1

DJ2LS, 2024-03-07:
Mh, this should have been improved with this version.
You are running the Windows installer, right?

On first start after installation FreeDATA needs some seconds, maybe waiting ten seconds, does the problem still exist?

pepefrog1234, 2024-03-07:
My computer is running windows 10. I reopened the software and waited, the problem still exists.

DJ2LS, 2024-03-07:
mh, this should occur, 

@Mashintime @dk5sm do you have the time, reproducing this? 

@pepefrog1234 is this the first time running FreeDATA?

pepefrog1234, 2024-03-07:
This is not the first time I have opened the program

DJ2LS, 2024-03-09:
@pepefrog1234 did you start the server? I see the network connection is red

dk5sm, 2024-03-09:
Please also check the modem port within config.ini; it should be 5000. Sometimes you can still find 3050 in it as a legacy.

pepefrog1234, 2024-03-10:
> connection
No I didn't start the server, is that the problem?

pepefrog1234, 2024-03-10:
> Please also check the modem port within config.ini; it should be 5000. Sometimes you can still find 3050 in it as a legacy. 

Okay, I'll check.

DJ2LS, 2024-03-10:
Yea I think so. But the latest installer should provide everything and you just have tk start the gui and the server should start automatically

pepefrog1234, 2024-03-10:
> Yea I think so. But the latest installer should provide everything and you just have tk start the gui and the server should start automatically 

Thanks to the development team, I've solved it.

closed 2024-03-10 by DJ2LS