DJ2LS/FreeDATA #705: retention of MHEARD

SpudGunMan, 2024-04-14:
**Is your feature request related to a problem? Please describe.**
keep the MHEARD list to share and maintain between restart

**Describe the solution you'd like**
MHEARD list to share with others or use in mesh routing

**Additional context**
a way to also clear it

DJ2LS, 2024-04-17:
FYI: Mesh is not yet functional in new server, we tested it as a proof of concept.  

The heard station list is also the data source for mesh routes. This means it will be automatically shared.

DJ2LS, 2024-07-27:
Still the open task with saving the heard stations list. 

@LA3QMA @Mashintime @dk5sm @deejdeej any ideas regarding this? Does it make sense saving the heard stations? We would have to store it locally in a file I think. But thats something I want to avoid, so we only have the database and config file as file related dependency. I also want to avoid saving the heard station information to the database as I not know how it will perform, when we have larger data sets stored... 🤔