DJ2LS/FreeDATA #706: Supported codec2 versions

sarcasticadmin, 2024-04-15:
**Is your feature request related to a problem? Please describe.**

As of July 2023 codec2 now has 2 repos (old and new). Explained in more detail here: https://github.com/drowe67/codec2?tab=readme-ov-file#old-code-and-deprecated-features but its resulted in the following two repos:

- https://github.com/drowe67/codec2 - new and currently at version 1.2.0
- https://github.com/drowe67/codec2-dev - old and current at version 1.1.1

**Describe the solution you'd like**

Is it feasible for FreeDATA to support either version at the moment? I just wanted to get some guidance on which codec2 repo would be recommended and supported for use with FreeDATA.

Mashintime, 2024-04-16:
Version 1.2 is what we are currently targetting.  I don't see it being feasible to stay compatible with 1.1.1 due to the speed of innovation of newer features being introduced in both codec2 and FreeData currently.

sarcasticadmin, 2024-04-16:
@Mashintime ok great, thats what I needed to know. Thank you!

closed 2024-04-16 by sarcasticadmin