DJ2LS/FreeDATA #710: Map widget

SpudGunMan, 2024-04-18:
An explorer like map widget displaying nodes without internet (points with grid on map)

DJ2LS, 2024-04-18:
Thanks, @SpudGunMan I like this feature request, have been thinking about this as well many times. 
I'm stuck with finding a nice offline map which saves implementation time.

@Mashintime what do you think? As a start we could implement a simple map, showing heard stations for example. But core requirement in my opinion is a nice offline map, maybe a vector map or so, bot based on images 🤔

DJ2LS, 2024-09-11:
[image]

closed 2024-09-11 by DJ2LS