DJ2LS/FreeDATA #711: Messaging Improvement

jpronans, 2024-04-24:
**Is your feature request related to a problem? Please describe.**
Long term testing has shown that its possible for me to receive a beacon from someone else, but they can't hear me due to local their noise level.
  
At present, the only way to ascertain if they can possibly receive a message is to first ping them, if the ping is successful, then there is a possibility that the message may go through. Not always.

**Describe the solution you'd like**
I can quickly think of two possibly ways to deal with it
1)
Before a message attempt is made, a ping gets sent to the other station, if the received SNR from the remote is enough to sustain a message session then the session should proceed to be established

2)
If I receive the remote station at -4 (for example) and I make a message attempt.  Apply a penalty (1db for example). The second, message attempt will not start until the received SNR is up to -3.. the third at -2, the 4th at -1 etc etc.. 
If a SNR is received higher than or equal this 'threshold' then an automatic message send can take place.

DJ2LS, 2024-04-29:
I made some first adjustments regarding to this:

I reduced the connection attempts to 5, and we are only resending the message, if we have a minimum SNR of -2dB.

DJ2LS, 2024-08-03:
I also added a slot based busy detection, sure, might need some fine tuning, but this might improve this topic as well. After some testing, I think we can close this one, as we have SNR based check and QRM based checks

closed 2024-09-08 by DJ2LS