DJ2LS/FreeDATA #718: Doesn't Recognize com0com Ports

KhunDoug, 2024-05-05:
This is an Icom 7800 using Win4IcomSuite and com0com virtual COM ports. Rigctl and Rigctld don't work properly with Win4IcomSuite. But CAT7200 implements PTT via RTS/DTR. A virtual COM port is assigned to CAT7200. I know CAT7200 works as I use it with other apps. FREEDATA does not recognize any of the com0com virtual ports. The only COM ports listed are the physical COM ports. The only way I can use FREEDATA is to close Win4IcomSuite and then setup Rigctld to use the physical COM port for the rig.

**Expected behavior**
I have multiple virtual COM ports. One of these is assigned to COM7200 to implement PTT via RTS/DTR. But the only COM ports that are listed are physical COM ports. That won't work with my rig, so COM7200 implements it via a virtual COM port.

**Logfiles**
https://wiki.freedata.app/en/usage/logging

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

- OS: Windows 11
- Latest Alpha build
- Windows

**Additional context**
I am using rev 2 of com0com. The rev 3 version will not operate correctly on this install of Windows 11. Rigctl and Rigctld do not work with Win4IcomSuite. Using CAT7200 works well to implement PTT via RTS/DTR.

DJ2LS, 2024-05-14:
Sorry for the late reply. 

I'm not able reproducing this because I don't have windows systems running. I did some research and it seems it's a problem related to the library we are using. There is already an open pull request: 
https://github.com/pyserial/pyserial/pull/70

But it's not being merged since years for some reason.

KhunDoug, 2024-05-14:
I can gather screen snapshots of the registry if that will help. Since I have real case scenario, just let me know what info you want collected and I will gather and submit.

DJ2LS, 2024-06-02:
There might be a possible fix for this - I think we can implement a custom module, at least its a hack, but it might add support for this.

closed 2025-04-05 by DJ2LS