DJ2LS/FreeDATA #734: [Bug]: FreeDATA Server fails to control rigctld properly if rigctld was invoked with the `--vfo` option

k8wu, 2024-05-26:
### FreeDATA Version

0.15.9-alpha

### What operating system are you using?

Windows

### Operating System Version

Windows 11 23H2

### Architecture

x64

### Area of problem

Rig Control

### Describe the bug

FreeDATA server fails to control a rigctld that was spawned from elsewhere if rigctld was started with VFO mode enabled (e.g. `rigctld -o` or `rigctld --vfo`). In this mode, rigctld requires the VFO to be specified in addition to the command and its parameter(s). For example:
- Normal command sent to rigctld to change frequency to 14.093 MHz: `F 14093000`
- The same command if VFO mode is enabled: `F A 14093000`

I don't recall if "A" will be the correct VFO to specify for all radios, but you can use "currVFO" to target whichever VFO the radio has currently active, and that would probably work for most cases.

### To Reproduce

Go to Settings -> Rig Control, then configure the program to connect to an instance of rigctld that is running with VFO mode enabled

### Expected behavior

The program should ideally be able to detect whether rigctld is running with VFO mode enabled, i.e. if an error is encountered when any command is sent to it without specifying the VFO or "currVFO", and then adjust any further commands accordingly.

DJ2LS, 2024-05-31:
....ready for testing in "develop branch" - it will be part of 0.16.0 release.

DJ2LS, 2024-08-17:
2024-08-17 19:43:36 [warning  ] Error getting ALC: could not convert string to float: 'USB\n2400'

closed 2024-09-08 by DJ2LS