DJ2LS/FreeDATA #736: [Bug]: Problems with recent develop version and rigctld

frspin, 2024-06-01:
### FreeDATA Version

Develop branch from Github

### What operating system are you using?

Ubuntu

### Operating System Version

Ubuntu 22.04

### Architecture

x64

### Area of problem

Server - Message System

### Describe the bug

Recent version (yesterday) from Github develop branch have problems with rigctld.
I use a shell proc for starting rigctld, freedata_server and freedata_gui
After starting freedata_server I get this errors:

Starting server with path: /mnt/Prog/server/freedata-server
Failed to start server process: Error: spawn /mnt/Prog/server/freedata-server ENOENT
    at ChildProcess._handle.onexit (node:internal/child_process:284:19)
    at onErrorNT (node:internal/child_process:477:16)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21) {
  errno: -2,
  code: 'ENOENT',
  syscall: 'spawn /mnt/Prog/server/freedata-server',
  path: '/mnt/Prog/server/freedata-server',
  spawnargs: []
}

and, after this:

2024-06-01 12:24.53 Error getting ALC: could not convert string to float: 'PKTUSB\n0'
2024-06-01 12:24.54 [RIGCTLD] Error getting TUNER state: invalid literal for int() with base 10: '0.000000'
2024-06-01 12:24.55 [RIGCTLD] Connected to rigctld at 127.0.0.1:4532
2024-06-01 12:24.55 Error getting ALC: could not convert string to float: 'PKTUSB\n0'

After this, and after Ctrl C to terminate, there is this error message:

2024-06-01 12:25.09 [RIGCTLD] Failed to connect to rigctld: [Errno 111] Connection refused
 
and a kill -9 is required for terminating server.

What is changed?

Regards
Franco Spinelli
IW2DHW

### To Reproduce

Start freedata_modem will  suffice for getting error.

### Expected behavior

Until yesterday all was working

frspin, 2024-06-01:
Another info:

adding --vfo to rigctld solve the problem so I think that in processing --vfo flagt there is some trouble, probably in 8f6bce9 commit.

Franco Spinelli
IW2DHW

DJ2LS, 2024-06-01:
Good to know, that’s the actual work in progress - could you do me the favour and checking which commands of rigctld are failing? This would help me, I can continue with this work tomorrow.Von meinem iPhone gesendet

frspin, 2024-06-01:
Il 01/06/24 13:36, DJ2LS ha scritto:
> Good to know, that’s the actual work in progress - could you do me the 
> favour and checking which commands of rigctld are failing? This would 
> help me, I can continue with this work tomorrow.

The problematic command seem to be getting ALC or TUNER state as for 
error message in my report.

For now I have added --vfo to rigctld invocation.

Franco Spinelli
IW2DHW

DJ2LS, 2024-06-02:
I think I've fixed this with latest commit to "develop" - if more problems occur, please wirte this to #733 as comment

closed 2024-06-02 by DJ2LS