DJ2LS/FreeDATA #760: [Feature Request]: Resume transfer if the connection fails

ND2W, 2024-08-07:
### Problem Description

If a transfer fails we have to start over, in some cases losing several hours if large files were attached. This could be caused by worsened propagation or a technical issue. A "resume transfer" feature could save significant time, which is especially important when sending critical data. This feature should apply to both text messages and files.

### Proposed Solution

FD should keep a log for each transfer detailing which data packets have been transmitted and acknowledged by the receiving station. Then it should be possible to read the log upon resuming, identify which data packets were not acknowledged, and resume from there.

This would require the receiving station to save the data packets it has already received.

### Alternatives Considered

N/A

DJ2LS, 2024-08-10:
this might work:

- create a session id by creating a checksum of the entire transmission data
- if a transmission has been failed, don't remove it from session list
- if a transmission with the same session id happens again AND we have the same callsign information, check if there is a failed transmission ( false positive checksums might occur ), if callsign differs, create new random id
- if one is available,IRS has to provide the buffer location
- ISS has to continue with the given buffer location
- If transmission is successful, remove it from session list

DJ2LS, 2024-08-29:
```    self.template_list[FR_TYPE.ARQ_SESSION_INFO_ACK.value] = {
            "frame_length": self.LENGTH_SIG0_FRAME,
            "session_id": 1,
            "total_crc": 4,
            "snr": 1,
            "speed_level": 1,
            "frames_per_burst": 1,
            "flag": 1,
        }```

We have to change total crc to offset for transmitting the data offset. Maybe we should also set a flag "resume". 

For validation, we have to check for the total checksum. If both are equal in memory, we can ensure we are resuming the correct data. Because of the fact, we are only using a 8bit int as session id, false positive night occur. So we have to ensure, deleting sessions after a transmission succeeds.

DJ2LS, 2024-08-30:
First working version - needs to be tested now

ND2W, 2024-08-30:
"Resume transfer" feature works as tested by N1QM and myself. Thanks for the implementation!

One suggestion: The timeout on the receiving side could be reduced from currently 2(?) min to 1 min. Otherwise the sending station will not be able to resume the transfer, as long as the receiver waits for the first (aborted) transmission to continue. 

The receiver has to timeout or manually hit the "stop" button before the sender can establish a connection again.

closed 2024-09-08 by DJ2LS