DJ2LS/FreeDATA #761: [Feature Request]: GUI STOP button should immediately abort transfer

ND2W, 2024-08-08:
### Problem Description

The STOP button in the GUI does not end the transfer right away. Instead FD tries to inform the other station that the transfer was stopped by the sending stations operator. FD will continue sending and sometimes it appears as if FD and the radio get stuck in transmit mode which could potentially damage the radio. The only way to stop it is to shut down the FD server.

This behavior can be problematic when one has to shut down the rig immediately. Also, if the path disappeared it will be impossible to let the other station know that the transfer was aborted.

### Proposed Solution

As soon as the STOP button is pressed, FD should immediately stop sending. The other station will eventually timeout and no longer wait for new packets. The timeout period on the receiving stations end could be reduced to a reasonable amount (e.g. 20 seconds).

In addition, the STOP button should show a pop-up that asks the user if they really want to abort the transfer. This will prevent a user from accidentially aborting their transfer if they happen to click on the STOP button by mistake.

### Alternatives Considered

N/A

DJ2LS, 2024-08-10:
Suggestion:
The button opens a popup, requesting additional check, then stopping transmission

DJ2LS, 2024-08-10:
@LA3QMA @deejdeej @Mashintime @dk5sm your opinion regarding to this?

LA3QMA, 2024-08-10:
I see the need to stop a transmission. Either to stop occupy the frequency or if there are an issue with the radio/swr etc.

So if its not much extra work adding a popup to verify that you want to terminate I would say this should be implemented.

But i would suggest that FD transmit ONE package if the channel is free before going QRT to the remote station telling it that we have stopped transmission. Then if the receiving station get this it would also stop. If not its just going to send retries. 

So yes to a popup to verify the STOP but send one "STOP" packet to the remote station.

deejdeej, 2024-08-10:
"As soon as the STOP button is pressed, FD should immediately stop sending"  Yes, definitely.  I would not have a popup at all, just stop transmission immediately.  If I'm pressing the stop button, it is very likely because there is an issue like a high SWR and I want the radio to stop NOW, not wait for a pop-up, and not send another transmission.  I consider "stop" to be essentially "emergency stop".

DJ2LS, 2024-08-10:
I implemented a first version in `develop`. At least it's a bit tricky, as we need two ways of stopping - one immediately and one with closing procedure. 

We might add a popup for informing the user, the "stop button" has been clicked, also informing then about a possible delay before a connection the same station can be established again as it has to timeout first.

closed 2024-09-08 by DJ2LS