DJ2LS/FreeDATA #762: [Bug]: First run after install the modem fails to start the selected audio device.

LA3QMA, 2024-08-09:
### FreeDATA Version

0.16.1-alpha

### What operating system are you using?

Other (specify below)

### Operating System Version

Windows and Linux

### Architecture

x86

### Area of problem

Installation

### Describe the bug

First time install network and radio control is ok. Modem the correct audio device is selected but fails to start.

Workaround is to close the modem check window, go to Settings, then fill in all required settings in each tab like callsign etc then in the "Modem" tab choose audio input/output. Now the modem starts.

After this the modem starts each time in the expected way.

### To Reproduce

First time instalation of FreeDATA

### Expected behavior

Modem should start

DJ2LS, 2024-08-18:
I'm not able reproducing this on Linux.

DJ2LS, 2024-11-10:
I reduced the priority of this one as it's not occurring that often it seems. Or at least  we don't get too much negative feedback. However, we should fix this.

DJ2LS, 2025-03-10:
I'm wondering about the status of this one

DJ2LS, 2025-03-10:
Does it still exist?

DJ2LS, 2025-05-06:
It seems this one isn't occurring anymore?

DJ2LS, 2025-07-14:
Is this issue still occurring

LA3QMA, 2025-07-15:
checked win/linux v0.17.6-beta and this is not an issue any more.

closed 2025-07-15 by LA3QMA