DJ2LS/FreeDATA #769: [Feature Request]: Show country label for callsigns and additional info in station info

DJ2LS, 2024-08-16:
### Problem Description

Actual callsigns are not clearly visible to which country they belong to

### Proposed Solution

Some nice country flags and an additional station info

### Alternatives Considered

None

DJ2LS, 2024-09-30:
@LA3QMA @Mashintime @dk5sm @deejdeej what do you think, is this something wich should be done by the logging software ( basic ADIF via UDP support has been added ), or should we have this implemented into the server? Challenge is, the corresponding information and logics need to be offline.

deejdeej, 2024-09-30:
Given that we can see their location on the map (both within the FD software (which I think already has country labels) and on the explorer), I'm not sure what additional information is being asked for here.

LA3QMA, 2024-09-30:
> Given that we can see their location on the map (both within the FD software (which I think already has country labels) and on the explorer), I'm not sure what additional information is being asked for here.

I agree. Extra info in the GUI even with a small icon or country name is going to use screen space.
The map GUI should show the country after a beacon. And even if it would be nice to have most logging software has this inbuilt.

Maybe later at some point a "hover over the callsign could show the country". Or let the sqlite have a country table and auto insert this when receiving a beacon. But i rather wait with this and focus on other functions for now.

closed 2024-09-30 by DJ2LS