DJ2LS/FreeDATA #773: [Bug]: audio loss when dis/connecting via RDP (Windows)

DJ2LS, 2024-08-17:
### FreeDATA Version

0.16.1.alpha

### What operating system are you using?

Windows

### Operating System Version

Windows 10

### Architecture

x86

### Area of problem

Server - Message System

### Describe the bug

When connecting to my Windows test machine with a running FreeDATA server via RDP, the waterfall freezes because the audio device reports an "input overflow". Also happens when disconnecting the RDP session:
image.png
I suspect that this is a (new) RDP bug; perhaps the audio device is briefly "touched" when disconnecting or connecting and thus snatched from the server. Maybe there is a chance to find a workaround for this and make FD a bit more resilient against this issue?!
‌
(Btw.: VARA seems to have the same issue, and shows up “frozen” after a connect, but recovers after a short time)

### To Reproduce

Connect via RDP to Windows instance

### Expected behavior

Audio crash

### Screenshots

[image]

DJ2LS, 2024-11-10:
@dk5sm does this Problem still exist?

dk5sm, 2024-11-10:
Yes, it´s still existing with 0.16.8:

`2024-11-10 13:50:51 [warning  ] [AUDIO STATUS]                 frames=4800 status=<sounddevice.CallbackFlags: input overflow> time=<cdata 'struct PaStreamCallbackTimeInfo *' 0x000000AF0A9DF6A8>`

As ist seems to a Windows/RDP related issue to me, you may close this report.
If you want to work around this issue, maybe a kind of watchdog which restarts the audiodevices may do the trick when an _input overflow_ occurs.

DJ2LS, 2025-03-10:
any news regarding this?

Mashintime, 2025-03-10:
I experience this is as well, and believe it's related to audio forwarding.  If it't enabled it'll stop working for me.  When disabled there's no issues.  Don't think this is something that can be addressed by us.

dk5sm, 2025-03-11:
Error still occurs with 0.16.11 on Windows 11 with the same warning message (see Nov 10 entry in this issue). It´s for sure a Windows-thing, but maybe there´s a way to work around this, as VARA seems to suffer from this issue as well. Watchdog -> restart audio part?

DJ2LS, 2025-03-11:
Only option would be a “healing mechanism”. Means if audio stops we could try restarting it somehow. Not sure if this is really needed. Otherwise we should write a wiki entry and closing this issue Von meinem iPhone gesendet