DJ2LS/FreeDATA #778: [Feature Request]: Tune function for antenna tuners

deejdeej, 2024-08-17:
### Problem Description

Add a second tune button, or make the existing tune button user configurable to send out a solid tone that can be used to tune an antenna tuner.  First press activates transmits the tune tone, and it stays transmitting until pressed a second time to stop the tune transmission.  This allows plenty of time for an external antenna tuner to process and select the best SWR.

### Proposed Solution

The current tune button does not stay transmitting long enough for some antenna tuners to work.  A solid tune tone seems to work better on some antenna tuners for matching SWR versus sending a data packet.

### Alternatives Considered

n/a

DJ2LS, 2024-09-30:
I'm not that happy with the "start stop" thing as it feels, it might lead to problems. Case: Tune is pressed, then we have high interference on computer because of tuning process, which maybe leads to a not accessible server. 

What about pressing "tune" with an adjustable timer, so Tune stays in tune for 5, 10, 15, 20 seconds, then tune will be released automatically.

deejdeej, 2024-09-30:
What about pressing once to start the tune, pressing a second time to stop the tune, and have a maximum time that the tune function can stay active before shutting off automatically?  This would emulate exactly the way that WSJT-X does it which has worked well for a number of years.

DJ2LS, 2024-09-30:
ready for testing with latest develop, see Tune section

deejdeej, 2024-09-30:
The Transmit Sine Start/Stop and default of 30 seconds timeout works great!
I'm not sure the signal being sent out is quite right, though.
If I set my rig to 100 watts, and click the test-frame transmit for 5 seconds, I see 75 watts out on my meter which seems correct.
If I click the Transmit-Sine button, I only see 30 watts out.  I would expect to see 100 watts or close to it, unless I am misunderstanding something.

DJ2LS, 2024-10-01:
I have to check, I think we should combine the sine with the volume settings. So they are sticking together and maybe adjusting default settings.

DJ2LS, 2024-10-01:
sending sine now respects the audio level, I also increased volume per default ( +3dB ). So I think as a first start it's working and we can close this one. Audio level normalization is a bigger topic, as we also have to check the data modes. But for doing this I need much more time. 

In other words: I like to close this one, if you are fine with this, @deejdeej

closed 2024-10-01 by DJ2LS