DJ2LS/FreeDATA #779: [Bug]: Pressing ABORT throws error

OH2JRB, 2024-08-18:
### FreeDATA Version

0.16.2-alpha

### What operating system are you using?

Other Linux

### Operating System Version

Linux freedata-experience 6.1.0-23-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.99-1 (2024-07-15) x86_64 GNU/Linux

### Architecture

x64

### Area of problem

Server - File Transfer

### Describe the bug

```
    |   File "/root/FreeDATA/FreeDATA/freedata_server/server.py", line 312, in post_modem_send_raw_stop
    |     session.transmission_aborted()
    |   File "/root/FreeDATA/FreeDATA/freedata_server/arq_session_irs.py", line 315, in transmission_aborted
    |     True, self.id, self.dxcall, False, self.state.name, statistics=self.calculate_session_statistics(self.confirmed_bytes, self.total_length))
    |                                                                                                      ^^^^^^^^^^^^^^^^^^^^
    | AttributeError: 'ARQSessionIRS' object has no attribute 'confirmed_bytes'
```

### To Reproduce

Press ABORT while sending [large] file over RF.

### Expected behavior

Session abort.

DJ2LS, 2024-08-18:
Thanks., please do a git pull. I pushed a fix which should solve this.

OH2JRB, 2024-08-18:
```
2024-08-18 11:17:32 [info     ] [ARQSessionISS][id=89][state=ISS_State.ABORTING]: session aborted
2024-08-18 11:17:32 [info     ] [ARQSessionISS][id=89][state=ISS_State.ABORTING]: ARQSessionISS state change from ABORTING to ABORTED
2024-08-18 11:17:32 [debug    ] Event:                         ev={'type': 'arq', 'arq-transfer-outbound': {'session_id': 89, 'dxcall': '******-1', 'statistics': {'total_bytes': 4428, 'duration': 115.82359981536865, 'bytes_per_minute': 642, 'time_histogram': {0: '2024-08-18T11:15:53.250717', 1: '2024-08-18T11:16:00.250920', 2: '2024-08-18T11:16:07.249641', 3: '2024-08-18T11:16:13.350490', 4: '2024-08-18T11:16:27.743841', 5: '2024-08-18T11:16:34.744061', 6: '2024-08-18T11:16:40.751974', 7: '2024-08-18T11:16:55.144181', 8: '2024-08-18T11:17:02.150493', 9: '2024-08-18T11:17:25.653495'}, 'snr_histogram': {0: 11, 1: 2, 2: 3, 3: 3, 4: 3, 5: 1, 6: 3, 7: 3, 8: 2, 9: 0}, 'bpm_histogram': {0: 0, 1: 0, 2: 92, 3: 153, 4: 701, 5: 616, 6: 602, 7: 876, 8: 804, 9: 657}}, 'success': False, 'state': 'ABORTED', 'data': False}}
2024-08-18 11:17:32 [info     ] [API] POST                     response_code=200 url=http://localhost:5000/modem/stop_transmission
2024-08-18 11:17:32 [debug    ] [MDM] [demod_audio] freedata_server state mode=SIGNALLING_ACK rx_status=3 sync_flag=--ST
2024-08-18 11:17:32 [debug    ] [MDM] [demod_audio] freedata_server state mode=SIGNALLING_ACK rx_status=10 sync_flag=E-S-
```

OH2JRB, 2024-08-18:
Seems to do the work now. Closing issue.

closed 2024-08-18 by OH2JRB