DJ2LS/FreeDATA #780: [Feature Request]: Editable Commonly used frequencies

OH2JRB, 2024-08-18:
### Problem Description

Make Commonly used frequencies editable and save the work "somewhere" for later use i.e. recalling for use.

### Proposed Solution

User can edit and save list of commonly used frequencies.

### Alternatives Considered

-

### Additional Information

To support local calling frequencies - like OH (Finland) now active on 7058 kHz.

DJ2LS, 2024-09-18:
I think it will cost too much effort at cost of simplicity, if we make the frequency list editable. 

Any other suggestions? @LA3QMA @dk5sm @Mashintime @deejdeej

dk5sm, 2024-09-18:
Maybe we can keep it simple: What about a textfile-based 'frequencies.ini' within /freedata_server? Instead of hard-coding some fixed frequencies, this could be regulated via a simple file. This is solved in a similar way in VarAC, for example: one frequency per line. The file is then simply read in when the server is started.

deejdeej, 2024-09-18:
Alternatively, have the list of frequencies in the config.ini file?  That keeps all of the configuration options within a single file.

dk5sm, 2024-09-18:
Current format of VarAC´s 'VarAC_frequencies.conf' file:

14.105.000|20m - Main
7.105.000|40m - Main

The pipe symbol is a delimiter. It is followed by a description, which is then also displayed in the selection list. I think this is a good idea to adopt ;-)

DJ2LS, 2024-09-18:
If we decide, doing this on server side, I would prefer the existing config.ini 

This also reminds ne about a feature "scheduled frequencies"

However, I think we should have some standard frequencies which are interoperable over all ITU zones . When running mesh, it's not making sense running several mesh networks on one band.

dk5sm, 2024-09-18:
D'accord - also a good place!