DJ2LS/FreeDATA #784: [Feature Request]: compressing and resizing images

LA3QMA, 2024-08-23:
### Problem Description

reminder to do research and later implement automatic resize/compression when sending images.

### Proposed Solution

Maybe a option that ask if FD should resize/compress or send "as is".

### Alternatives Considered

Option in the configuration that globally enable/disable automatic resize/compression.

DJ2LS, 2024-09-17:
@LA3QMA ready for testing

DJ2LS, 2024-09-18:
@LA3QMA hows the image size? Should it be smaller in size? 
500x500px seems to be too much while 250x250px seems to be too small

DJ2LS, 2024-09-18:
Okay, I increased this to 750px, less doesn't make that much sense as the images become too small. However, we have good compression results. But it should be clear, that sending images requires  a good channel and lot of data needs to move over the ether.

dk5sm, 2024-09-19:
[image]
[image]

I encountered some strange behaviour while using latest FD version and fiddling around with image compression:

1. The attached file 3percent_20240906_140344.jp is 121 x 68 px in size and results in:
[image]

2. The attached file heise.png is 16 x 16 px (336 bytes!) and results in:
[image]

The former can perhaps be explained with overhead, but the latter is weird....

DJ2LS, 2024-09-19:
pushed a commit, please do a git pull and npm run build 
should be fixed now

dk5sm, 2024-09-19:
Looks good now! The 226 bytes pic expands to 458 bytes, which is probably due to the normal overhead. The other one shrinks from 6915 to 2179 bytes.

closed 2024-09-28 by DJ2LS