DJ2LS/FreeDATA #798: [Bug]:Linux install script should test for npm

danieldamon, 2024-09-11:
### FreeDATA Version

non development Sept 24

### What operating system are you using?

Ubuntu

### Operating System Version

ubuntu 22.04.4

### Architecture

x86

### Area of problem

Installation

### Describe the bug

npm was not installed on my machine.  I modified the install script as shown by the diff file:
340,345d339
< 
< if ! command -v npm &> /dev/null
< then
<     sudo apt install -y npm
<     exit 1
< fi

### To Reproduce

remove npm and re install

### Expected behavior

It told me npm was missing

### Screenshots

none

### Log Files

none

DJ2LS, 2024-09-11:
@deejdeej  do you have the time, having a look at this?

@danieldamon thanks for the feedback!

deejdeej, 2024-09-12:
We don't use the distro (apt) supplied version of nodejs (npm) because it can often be too old.  The script downloads and installs nodejs from their site into ~/.nvm in the home directory, and installs the packages we need into that location.  

If the script gives an error, "Something went wrong.  npm install.sh not downloaded", then there was an issue with downloading the nodejs software.  All the script really can do in this case is tell you and fail.

The relevant commands start at line 126 in the install script if you are curious about what/how it is doing.

danieldamon, 2024-09-13:
On the discord, it said that you were going to a different distribution method. Perhaps not using scripts any more?  Maybe I should have installed the zip file?  Feel free to contact me to look at anything.

deejdeej, 2024-09-13:
We moved to the script installs for Linux systems.  Previously it was a really messy manual install method for Linux.  Current instructions for Linux can be found at: 
https://wiki.freedata.app/en/installation/linux-install-script

What happens when you run "bash install-freedata-linux.sh"  using an unmodified script?

danieldamon, 2024-09-13:
npm -version reports version 8.5.1 if that helps

danieldamon, 2024-09-13:
Earlier, you mentioned that you don't use the distro npm, so I uninstalled it and now I get the following error:

Downloading regular version
Cloning into 'FreeDATA'...
remote: Enumerating objects: 29750, done.
remote: Counting objects: 100% (2938/2938), done.
remote: Compressing objects: 100% (1004/1004), done.
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 4024 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
*************************************************************************
Changing Directory into FreeDATA
*************************************************************************
Something went wrong.  FreeDATA software not downloaded from git.

danieldamon, 2024-09-13:
maybe just a bad connection because i ran the git command manually with success.  Will try finishing the script and see what happens.

danieldamon, 2024-09-13:
When I continue running, the failure indicates that npm is not installed, which is why I made the original changes:

*************************************************************************
Building the FreeDATA GUI frontend
*************************************************************************
install2.sh: line 70: npm: command not found
install2.sh: line 73: npm: command not found

danieldamon, 2024-09-13:
Sorry line numbers in the script are incorrect.  It's because I cut the script after the git command and restarted it.

danieldamon, 2024-09-13:
After installing npm (which was not originally in the distribution), the error is:

 FreeDATA@0.16.3-alpha build
> vue-cli-service build

All browser targets in the browserslist configuration have supported ES module.
Therefore we don't build two separate bundles for differential loading.

✔  Building for production...
 ERROR  Error: @vitejs/plugin-vue requires vue (>=3.2.13) or @vue/compiler-sfc to be present in the dependency tree.
Error: @vitejs/plugin-vue requires vue (>=3.2.13) or @vue/compiler-sfc to be present in the dependency tree.
    at Object.<anonymous> (/freedata/FreeDATA/freedata_gui/node_modules/vue-loader/dist/compiler.js:14:15)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:85:18)
    at Object.<anonymous> (/freedata/FreeDATA/freedata_gui/node_modules/vue-loader/dist/index.js:29:20)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:85:18)
    at /freedata/FreeDATA/freedata_gui/node_modules/@vue/cli-service/lib/config/base.js:154:16
    at /freedata/FreeDATA/freedata_gui/node_modules/@vue/cli-service/lib/Service.js:268:40
    at Array.forEach (<anonymous>)
    at Service.resolveChainableWebpackConfig (/freedata/FreeDATA/freedata_gui/node_modules/@vue/cli-service/lib/Service.js:268:26)
    at PluginAPI.resolveChainableWebpackConfig (/freedata/FreeDATA/freedata_gui/node_modules/@vue/cli-service/lib/PluginAPI.js:145:25)
    at module.exports (/freedata/FreeDATA/freedata_gui/node_modules/@vue/cli-service/lib/commands/build/resolveAppConfig.js:9:22)
    at build (/freedata/FreeDATA/freedata_gui/node_modules/@vue/cli-service/lib/commands/build/index.js:147:50)
    at /freedata/FreeDATA/freedata_gui/node_modules/@vue/cli-service/lib/commands/build/index.js:69:13
    at Service.run (/freedata/FreeDATA/freedata_gui/node_modules/@vue/cli-service/lib/Service.js:262:12)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
dan@dan-HP-Laptop-17z-ca100:/freedata/FreeDATA/freedata_gui$

deejdeej, 2024-09-14:
Out of curiousity, are you running the installer exactly as shown on the wiki, including the "bash" command before the script?

Please do so if you haven't been: "bash install-freedata-linux.sh"

I just tried on an Ubuntu 22.04.04 test system that I have here, and the script ran to completion with no issues, installed npm, the git repo, etc.

If you are having network issues that are affecting the ability to download the git repo and nodejs files from the Internet, that is a very likely cause of the issues you are seeing.

I did just add an update to the install script to handle OS version checking a bit better, so when you update your Ubuntu 22.04.04 system to 22.04.05 it should handle that change.

danieldamon, 2024-09-14:
Yes, I ran it exactly as described.  Unfortunately, my internet is very slow, so that may explain some failures in git clone. However, I am sure that I did not have npm installed.  If your test machine has npm installed, it is probably because someone installed it as some point in this process. same with vue.

Thanks for your hard work.  I'm just trying to get this all working better because it seems like an excellent project.

danieldamon, 2024-09-14:
something fishy in the install script. Looks like it installs nvm, then says it is version npm -v

deejdeej, 2024-09-14:
My test machine does not have npm on it to start.  The script installs npm as part of the nvm/node installation that it performs.

If nvm and node are successfully installed by the script, it prints out two lines similar to:
nvm is version 10.8.2
node is version v20.17.0

This is normal and expected.

danieldamon, 2024-09-14:
I see the install script has been modified slightly.  I'll try again.  I have not removed nvm in past attempts. I wonder if that matters?

danieldamon, 2024-09-14:
Building the FreeDATA GUI frontend
*************************************************************************
install-freedata-linux.sh: line 343: npm: command not found
install-freedata-linux.sh: line 346: npm: command not found
Same thing.  Next, I'll try removing both nvm and npm

danieldamon, 2024-09-14:
Still no luck. Could not uninstall nvm, but instructions said to clear ~/.nvm 

Here is from your script:  I don't see where npm is installed.  But the last line reports that nvm version is the npm version, clearly incorrect:
    132 wget https://raw.githubusercontent.com/nvm-sh/nvm/v0.39.7/install.sh
    133 if [ -f "install.sh" ];
    134 then
    135         XDG_CONFIG_HOME=""
    136         chmod 750 install.sh
    137         ./install.sh
    138 else
    139         echo "Something went wrong.  npm install.sh not downloaded."
    140         exit 1
    141 fi
    142 
    143 if [ -f "$HOME/.nvm/bash_completion" ];
    144 then
    145         export NVM_DIR="$HOME/.nvm"
    146         [ -s "$NVM_DIR/nvm.sh" ] && \. "$NVM_DIR/nvm.sh"
    147         [ -s "$NVM_DIR/bash_completion" ] && \. "$NVM_DIR/bash_completion"
    148         nvm install 20
    149         echo "nvm is version" `npm -v`

deejdeej, 2024-09-14:
I guess I do not know how to help you.  I would encourage you to ask on the Discord and maybe one of the other folks will have better luck.

I have corrected the typo in the echo statement.

danieldamon, 2024-09-15:
I can download that install.sh script an run it manually. It does not install npm.  here is the output:

bash install.sh
=> nvm is already installed in /home/dan/.nvm, trying to update using git
=> => Compressing and cleaning up git repository

=> nvm source string already in /home/dan/.bashrc
=> bash_completion source string already in /home/dan/.bashrc
=> Close and reopen your terminal to start using nvm or run the following to use it now:

export NVM_DIR="$HOME/.nvm"
[ -s "$NVM_DIR/nvm.sh" ] && \. "$NVM_DIR/nvm.sh"  # This loads nvm
[ -s "$NVM_DIR/bash_completion" ] && \. "$NVM_DIR/bash_completion"  # This loads nvm bash_completion

danieldamon, 2024-09-15:
Please do me a favor. In your test machine, run

sudo apt-get remove npm

Then try running the install script again.

Thanks

danieldamon, 2024-09-15:
when I try running:

nvm install node

I get the following error:
Downloading and installing node v20.17.0...
Downloading https://nodejs.org/dist/v20.17.0/node-v20.17.0-linux-x64.tar.xz...
Warning: Failed to open the file                                               
Warning: /home/dan/.nvm/.cache/bin/node-v20.17.0-linux-x64/node-v20.17.0-linux-
Warning: x64.tar.xz: Permission denied
curl: (23) Failure writing output to destination

Binary download from https://nodejs.org/dist/v20.17.0/node-v20.17.0-linux-x64.tar.xz failed, trying source.
grep: /home/dan/.nvm/.cache/bin/node-v20.17.0-linux-x64/node-v20.17.0-linux-x64.tar.xz: No such file or directory

danieldamon, 2024-09-15:
Fortunately, I found the solution online at https://askubuntu.com/questions/1385123/nvm-cant-install-node

,,,

I tried all suggestions here, but that didn't do it. I tried rebooting, and even downloading that file and placing it in nvm cache bin.. no luck until..
Simple Solution:

if you have installed curl through using snap then uninstall that first:

sudo snap remove curl

then, re-install by using apt instead with:

sudo apt install curl

and now you should be fine to install node with:

npm install 18

(v18 at the moment, replace 18 with the version you need)

danieldamon, 2024-09-15:
I'm in the process of installing, and I think it will probably work. But it still is a problem for you. Many users have curl installed using snap. I suggest you execute:

snap list | grep curl

and if curl is installed by snap, advise the user to remove it.

In my case, I did not reinstall with apt-get, figuring your script would install it

danieldamon, 2024-09-15:
Success!  Hope you can handle the snap installed curl issue

deejdeej, 2024-09-15:
Nice find!  I'm glad you were able to solve it.  

Neither one of my Ubuntu test machines (22.04, 24.04) has curl installed (via snap or apt).  These are "fresh" installs setup for the sole purpose of testing FreeDATA installations, and since we don't use curl for FreeDATA we don't do anything with it.

It is impossible for us to cover every change someone might have done to their system, but this one might at least warrant a comment in the Wiki if we don't decide to detect this in the script.

Thank you for finding it.

danieldamon, 2024-09-19:
I hope you do document this issue because it makes it impossible to install
Freedata. I'd guesstimate that as much as 10% of linux users use curl
installed with snap.
Thanks

DJ2LS, 2024-09-20:
I‘m wondering if we could solve this easily with „if else“, checking if the corresponding tool is available.Von meinem iPhone gesendet

DJ2LS, 2024-09-20:
@LA3QMA could you add this to the wiki as an information?

DJ2LS, 2024-09-20:
I reopened this. I suggest we are closing it if we documented it and/or found a final solution.

LA3QMA, 2024-09-20:
> @LA3QMA could you add this to the wiki as an information?

wiki is updated with the following info:
Requirements: Ubuntu like 22.04, 24.04 is not pre-installed with curl. For now this has to manually be installed before using the install script. We are working on updating the script to handle this automatically if possible. You can use snap or apt to install CURL.

deejdeej, 2024-09-20:
@LA3QMA We don't need curl to be installed for the FD install script.  The FreeDATA installation does not use curl. The problem is that nvm fails to install properly if curl is installed via snap, so the Wiki should state something to the effect of 'Curl installed via snap has a bug that prevents the NVM software from being installed correctly.  If you have installed curl using snap, please remove it by running "sudo snap remove curl".  If you need curl on your system, please install it via apt with "sudo apt install curl".'

LA3QMA, 2024-09-20:
> @LA3QMA We don't need curl to be installed for the FD install script. The FreeDATA installation does not use curl. The problem is that nvm fails to install properly if curl is installed via snap, so the Wiki should state something to the effect of 'Curl installed via snap has a bug that prevents the NVM software from being installed correctly. If you have installed curl using snap, please remove it by running "sudo snap remove curl". If you need curl on your system, please install it via apt with "sudo apt install curl".'
> 
> I would put this under Knowledgebase, Linux as a separate item rather than as a warning on the Install script page. I suspect this is going to affect only a very few people.

Ahh qsl. I'll fix this. tnx

deejdeej, 2024-09-20:
@LA3QMA Looks good!  Thank you for adding this!

closed 2024-09-20 by deejdeej