DJ2LS/FreeDATA #804: [Bug]: Rigctl not founs

danieldamon, 2024-09-13:
### FreeDATA Version

v0.16.2-alpha

### What operating system are you using?

Windows

### Operating System Version

windows 11 32H2 AMD Ryzen

### Architecture

x64

### Area of problem

Rig Control

### Describe the bug

2024-09-12 17:59:12 [info     ] [C2 ] Libcodec2 loaded         path=C:\Users\dangd\AppData\Local\FreeDATA\FREEDA~1\lib\codec2\libcodec2.dll
Using config from C:\Users\dangd\AppData\Local\FreeDATA\FREEDA~1\config.ini
2024-09-12 17:59:13 [info     ] [DatabaseManager]: Initialized database
2024-09-12 17:59:13 [info     ] [DatabaseManager]: Database vacuumed successfully
Hamlib args ==>['-m', '2028', '-r', 'COM3', '-s', '115200', '-p', 'COM3', '-P', 'USB', '-D', 'RIG', '--set-conf', 'dtr_state=ON', '--set-conf', 'data_bits=8', '--set-conf', 'stop_bits=1', '']
2024-09-12 17:59:13 [warning  ] Rigctld binary not found.
2024-09-12 17:59:13 [info     ] [DatabaseManager]: Database reindexed successfully
2024-09-12 17:59:13 [info     ] [DatabaseManager]: Database integrity check passed
2024-09-12 17:59:13 [info     ] ---------------------------------------------------
2024-09-12 17:59:13 [info     ]
2024-09-12 17:59:13 [info     ] [GUI] AVAILABLE ON http://127.0.0.1:5001/gui
2024-09-12 17:59:13 [info     ] just open it in your browser
2024-09-12 17:59:13 [info     ]
2024-09-12 17:59:13 [info     ] ---------------------------------------------------
2024-09-12 17:59:15 [warning  ] [RIGCTLD] Failed to connect to rigctld: [WinError 10061] No connection could be made because the target machine actively refused it
2024-09-12 17:59:15 [info     ] tested audio devices           result=[True, True]
2024-09-12 17:59:15 [info     ] starting freedata_server....
2024-09-12 17:59:15 [debug    ] [C2 ] Create 48<->8 kHz resampler
2024-09-12 17:59:15 [debug    ] [C2 ] Creating audio buffer    size=9600
2024-09-12 17:59:15 [debug    ] [C2 ] Creating audio buffer    size=9600
2024-09-12 17:59:15 [debug    ] [C2 ] Creating audio buffer    size=9600
2024-09-12 17:59:15 [debug    ] [C2 ] Creating audio buffer    size=9600
2024-09-12 17:59:15 [debug    ] [C2 ] Creating audio buffer    size=9600
2024-09-12 17:59:15 [debug    ] [C2 ] Creating audio buffer    size=9600
2024-09-12 17:59:15 [debug    ] [C2 ] Creating audio buffer    size=9600
2024-09-12 17:59:15 [debug    ] [C2 ] Creating audio buffer    size=9600
2024-09-12 17:59:15 [info     ] loading frame dispatcher.....

2024-09-12 17:59:15 [debug    ] Event:                         ev={'freedata_server': 'started'}
2024-09-12 17:59:15 [info     ] [MDM] init: get audio devices  input_device=3cdc output_device=0129
2024-09-12 17:59:15 [info     ] [MDM] init: receiving audio from 'Stereo Mix (Realtek HD Audio Stereo input)'
2024-09-12 17:59:15 [info     ] [MDM] init: transmiting audio on 'Speakers 2 (Realtek HD Audio output with HAP)'
2024-09-12 17:59:15 [debug    ] [MDM] init: starting pyaudio callback and decoding threads
2024-09-12 17:59:15 [debug    ] [C2 ] Create 48<->8 kHz resampler
2024-09-12 17:59:15 [info     ] [DatabaseManagerBeacon]: Deleted 0 beacons older than 2 days
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=307 url=http://127.0.0.1:5001/gui
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/gui/
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/version
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/devices/audio
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/devices/serial
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/radio
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/modem/state
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/modem/state
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/version
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/devices/audio
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/radio
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/devices/serial
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/config
[... 6 lines cut ...]
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/config
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/freedata/messages
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/radio
{'NETWORK': {'modemaddress': '127.0.0.1', 'modemport': 5001}, 'STATION': {'mycall': 'KE0UR', 'mygrid': 'DM45gg', 'myssid': 3, 'ssid_list': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9], 'enable_explorer': False, 'enable_stats': False, 'respond_to_cq': True, 'enable_callsign_blacklist': False, 'callsign_blacklist': []}, 'AUDIO': {'input_device': '3cdc', 'output_device': '0129', 'rx_audio_level': 8, 'tx_audio_level': 8}, 'RIGCTLD': {'ip': '127.0.0.1', 'port': 4532, 'path': '', 'command': '', 'arguments': '', 'enable_vfo': True}, 'RADIO': {'control': 'rigctld_bundle', 'model_id': 2028, 'serial_port': 'COM3', 'serial_speed': 115200, 'data_bits': 8, 'stop_bits': 1, 'serial_handshake': 'None', 'ptt_port': 'COM3', 'ptt_type': 'USB', 'serial_dcd': 'RIG', 'serial_dtr': 'ON', 'serial_rts': 'OFF'}, 'TCI': {'tci_ip': '127.0.0.1', 'tci_port': 50001}, 'MESH': {'enable_protocol': False}, 'MODEM': {'enable_hmac': False, 'enable_morse_identifier': False, 'tx_delay': 50, 'maximum_bandwidth': 2438, 'enable_socket_interface': False}, 'SOCKET_INTERFACE': {'enable': False, 'host': '127.0.0.1', 'cmd_port': 8000, 'data_port': 8001}, 'MESSAGES': {'enable_auto_repeat': False}, 'GUI': {'auto_run_browser': True}}
2024-09-12 17:59:16 [info     ] [API] POST                     response_code=200 url=http://127.0.0.1:5001/config
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/config
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/radio
2024-09-12 17:59:16 [info     ] [API] GET                      response_code=200 url=http://127.0.0.1:5001/freedata/messages
{'NETWORK': {'modemaddress': '127.0.0.1', 'modemport': 5001}, 'STATION': {'mycall': 'KE0UR', 'mygrid': 'DM45gg', 'myssid': 3, 'ssid_list': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9], 'enable_explorer': False, 'enable_stats': False, 'respond_to_cq': True, 'enable_callsign_blacklist': False, 'callsign_blacklist': []}, 'AUDIO': {'input_device': '3cdc', 'output_device': '0129', 'rx_audio_level': 8, 'tx_audio_level': 8}, 'RIGCTLD': {'ip': '127.0.0.1', 'port': 4532, 'path': '', 'command': '', 'arguments': '', 'enable_vfo': True}, 'RADIO': {'control': 'rigctld_bundle', 'model_id': 2028, 'serial_port': 'COM3', 'serial_speed': 115200, 'data_bits': 8, 'stop_bits': 1, 'serial_handshake': 'None', 'ptt_port': 'COM3', 'ptt_type': 'USB', 'serial_dcd': 'RIG', 'serial_dtr': 'ON', 'serial_rts': 'OFF'}, 'TCI': {'tci_ip': '127.0.0.1', 'tci_port': 50001}, 'MESH': {'enable_protocol': False}, 'MODEM': {'enable_hmac': False, 'enable_morse_identifier': False, 'tx_delay': 50, 'maximum_bandwidth': 2438, 'enable_socket_interface': False}, 'SOCKET_INTERFACE': {'enable': False, 'host': '127.0.0.1', 'cmd_port': 8000, 'data_port': 8001}, 'MESSAGES': {'enable_auto_repeat': False}, 'GUI': {'auto_run_browser': True}}
2024-09-12 17:59:16 [info     ] [API] POST                     response_code=200 url=http://127.0.0.1:5001/config
{'NETWORK': {'modemaddress': '127.0.0.1', 'modemport': 5001}, 'STATION': {'mycall': 'KE0UR', 'mygrid': 'DM45gg', 'myssid': 3, 'ssid_list': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9], 'enable_explorer': False, 'enable_stats': False, 'respond_to_cq': True, 'enable_callsign_blacklist': False, 'callsign_blacklist': []}, 'AUDIO': {'input_device': '3cdc', 'output_device': '0129', 'rx_audio_level': 8, 'tx_audio_level': 8}, 'RIGCTLD': {'ip': '127.0.0.1', 'port': 4532, 'path': '', 'command': '', 'arguments': '', 'enable_vfo': True}, 'RADIO': {'control': 'rigctld_bundle', 'model_id': 2028, 'se [... cut ...]
2024-09-12 17:59:16 [info     ] [API] POST                     response_code=200 url=http://127.0.0.1:5001/config
2024-09-12 17:59:17 [warning  ] [RIGCTLD] Failed to connect to rigctld: [WinError 10061] No connection could be made because the target machine actively refused it

### To Reproduce

Use Hamlib internal.  

### Expected behavior

Should not get message that Rigctld could not be found. 

### Additional context

I'm using TRUSDX transciever which conforms to Kenwood TS480.  I have used it with WSTJX which also uses Hamlib, so this should work.

danieldamon, 2024-09-13:
First warning in the log may not be obvious. It is:

2024-09-12 17:59:13 [warning  ] Rigctld binary not found.

danieldamon, 2024-09-13:
so then I did npm i vue@3.2.13 to push the issue along and got the following error:

ERROR  Failed to compile with 1 error                               3:31:42 PM

 error  in ./src/main.js

Syntax Error: Thread Loader (Worker 0)
[BABEL] /freedata/FreeDATA/freedata_gui/src/main.js: Unexpected token '.' (While processing: "/freedata/FreeDATA/freedata_gui/node_modules/@vue/cli-plugin-babel/preset.js")

DJ2LS, 2024-09-14:
It might be possible, there is a problem with Hamlib and x64 systems. @dk5sm, maybe you can assist here if you have the time?

dk5sm, 2024-09-14:
@danieldamon I have had a (at least) similar issue here. To help with your case, plase allow some questions:

1. I assume that you are also using the FreeDATA Windows Installer package - correct?
2. What rigctl mode are you using? See Settings -> Rigcontrol: internal or external Hamlib?

As far as your log is concerned, I am missing the Hamlib path, which leads to the conclusion that you may have configured _external Hamlib_. Please switch to internal Hamlib and make sure that you have set the appropriate settings for your transceiver.

After restarting the server, a successful hamlib start should read like this:
```
Hamlib args ==>['-m', '3073', '-r', 'COM5', '-s', '115200', '-P', 'RIG', '--set-conf', 'data_bits=8', '--set-conf', 'stop_bits=1', '']
2024-09-14 10:18:48 [info     ] Attempting to start rigctld using binary found at: C:\Users\[Username]\AppData\Local\FreeData\freedata-server\lib\hamlib\hamlib-w64-4.5.5\bin\rigctld.exe

```

73 de
Stefan | DK5SM.

danieldamon, 2024-09-14:
Yes, I am using the windows .exe installer.

The mode I am using is Rigcontrol:internal.  I did not install an external hamlib

danieldamon, 2024-09-14:
Thanks for your work on this. I'd love to have something that works on all platforms.  Currently it looks like it is not used widely,, so I'm willing to test to help out.

danieldamon, 2024-09-15:
By the way, my computer is dual boot with Linux.  I just got it working in Linux, so apparently I'm doing it right.

dk5sm, 2024-09-15:
OK, please check whether these files are contained in this path (C:\Users\[Username]\....):

[image]

Check also with the file date: I had an issue with old/corrupt files. To be on the safe side, please unzip all the files you find in C:\Users\[Username]\AppData\Local\FreeData\freedata-server\lib\hamlib\hamlib-w64-4.5.5.zip, and then test them again.

danieldamon, 2024-09-15:
In fact, there is no bin directory, although that directory does exist in the zip archive. How is it extracted, and I will try manually to see what happens.

danieldamon, 2024-09-15:
In fact, it seems to work when I extract manually

DJ2LS, 2024-09-16:
Thanks for the feedback! So you are missing just the bin folder, right? Did this occur at first installation. Or as an update?

DJ2LS, 2024-09-17:
It might be a problem with nuitka bundling. Maybe the corresponding subfolders are ignored

DJ2LS, 2024-09-17:
confirming its a nuitka problem

DJ2LS, 2024-09-17:
@danieldamon As an additional information: If you are install hamlib system wide, it will also work. However, it seems I've found a fix

danieldamon, 2024-09-17:
Thanks! should I try again?

DJ2LS, 2024-09-17:
The windows package isn’t published yet, so we need to release it first.Von meinem iPhone gesendet

DJ2LS, 2024-09-18:
@LA3QMA could you Donne the favour and testing the internal development nsis package?

danieldamon, 2024-09-18:
Not sure I understand nsis, but am looking on github

danieldamon, 2024-09-18:
Sorry, I'll be on a plane most of the day, but I can continue Thursday.  NSIS looks like an open source windows installer.  Unfortunately, I am not familiar with it. Can you provide pointers?

DJ2LS, 2024-09-21:
It's the windows installer. We can test this with next release or depending on feedback from @LA3QMA

LA3QMA, 2024-09-21:
I installed this on win11. turned on my kx3 after connecting the cables and started FD. selected audio and internal rigctl usb port. then all green. then i go to Settings and select KX3. then CAT and rx/tx works.

Did a short test with the parameter "rigctld VFO parameter" on/off. Both seems to work on KX3.

closed 2024-09-28 by DJ2LS