DJ2LS/FreeDATA #806: [Bug]: RX/TX level adjust within 'tune' widget does not work

dk5sm, 2024-09-16:
### FreeDATA Version

0.16.4 (latest dev code as of 09/15)

### What operating system are you using?

Ubuntu

### Operating System Version

24.04.1

### Architecture

ARM Raspberry Pi

### Area of problem

GUI

### Describe the bug

Using the RX/TX level sliders within the 'Tune' widget has no effect. Only when using these controls within Settings -> Modem the Server restarts and the values are set.

### To Reproduce

1. click on 'Tune' widget
2. modify value for RX or TX
3. find that the server is not being restarted when ne value is set

### Expected behavior

Same behavior as within Settings/Modem.

DJ2LS, 2024-09-16:
OnChange not imported in main modals. Should be an easy fix

DJ2LS, 2024-09-17:
fixed, @dk5sm please test and confirm if its working. If so, please close the issue.

dk5sm, 2024-09-17:
confirm - fixed.

commit 9bac398e (2024-09-17): fix #806
closed 2024-09-17 by dk5sm