DJ2LS/FreeDATA #807: [Bug]: Strange issue with explorer

frspin, 2024-09-16:
### FreeDATA Version

0.16.4alpha

### What operating system are you using?

Ubuntu

### Operating System Version

24.04 full updated

### Architecture

x86

### Area of problem

GUI

### Describe the bug

I have a connection with  SQ9SVT-1 sending a message.
In FreeDATA Explorer in my station I can see SQ9SVT-1 as "Last heard" but in pop-up for SQ9SVT-1 there is not my call in "Last heard"
Is this correct?

### To Reproduce

Connect a station and see if your call is shown in Explorer

### Expected behavior

I expect that my call is shown in pop-up.

DJ2LS, 2024-09-17:
This might be a bug, but we need to await the 0.16.4 release and asking SQ9SVT updating as well. I fixed some things in this area. However, it might be possible, SQ9SVT didnt hear you and therefore didnt update its heard stations.

frspin, 2024-09-17:
Il 17/09/24 10:11, DJ2LS ha scritto:
> This might be a bug, but we need to await the 0.16.4 release and asking SQ9SVT updating as well. I 
> fixed some things in this area. However, it might be possible, SQ9SVT didnt hear you and therefore 
> didnt update its heard stations.

We have a message exchange so SQ9SVT have heard my signal.

Franco Spinelli
IW2DHW

frspin, 2024-09-19:
Another strange issue.

I send a CQ and receive a QRV from PA1ER-1. After this in Explorer pop-up for my call I can see PA1ER as received and in pop-up for PA1ER there is my call as received.

If I do a connection to PA1ER and send a message in my pop-up disappear PA1ER and in PAI1ER pop-up disappear my call.

Is this correct?

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2024-10-01:
Can you provide a screenshot? So it's then easier to check if its correct or now.

frspin, 2024-10-01:
Il 01/10/24 11:28, DJ2LS ha scritto:
> Can you provide a screenshot? So it's then easier to check if its correct or now.

I will do it as soon as I am able to be heard from someone with this problem!
On 14093 I am unable to be heard.

Franco Spinelli

frspin, 2024-10-01:
Il 01/10/24 11:28, DJ2LS ha scritto:
> Can you provide a screenshot? So it's then easier to check if its correct or now.
> 

Here 3 screenshots:

First is after callig CQ and receiving QRV from ON4FDN. In OF4FDN popup there is my station (13:27)
After this I have opened a chat with this station (13:29:20) and send a message
In latest screenshot (13:29:48) my call disappear from ON4FDN popup and also in my popup.

Regards

Franco Spinelli
IW2DHW

DJ2LS, 2024-10-01:
Franco, the screenshots are missing for some reason

DJ2LS, 2024-10-01:
The reason for this might be the case, where no grid square information is available. So this behavior is expected, but we also might find a better way for this somewhen. The explorer itself needs some adjustments, but I'm actually focusing on the modem stability. As soon as it is running fine, we can improve the explorer.

frspin, 2024-10-01:
Il 01/10/24 17:36, DJ2LS ha scritto:
> The reason for this might be the case, where no grid square information is available. So this
> behavior is expected, but we also might find a better way for this somewhen. The explorer itself
> needs [...]

Not a big problem.
But I have grid square set and also ON4DFN.
ON4DFN don't have frequency set. Can be this the problem?

Franco Spinelli
IW2DHW

DJ2LS, 2024-10-01:
I think it’s the fact we don’t exchange grid when sending data Von meinem iPhone gesendet

Not a big problem.
But I have grid square set and also ON4DFN.
ON4DFN don't have frequency set. Can be this the problem?

Franco Spinelli
IW2DHW

frspin, 2024-10-01:
Not true my last message.
ON4DFN have frequency set.
Is latest GUI version that show "NaN" as frequency.
Rigctld is correctly working and I can change frequency but GUI and FD Explorer show "NaN".

Loading http://localhost:5000/radio show  "err" as "radio_frequency" parameter.

Is this another bug or I am required to do some config changes?

Franco Spinelli
IW2DHW

DJ2LS, 2024-10-01:
Looks like the frequency isn't fetched correctly. I did some changes to rigctld today. Are you running latest code?

frspin, 2024-10-01:
Il 01/10/24 20:21, DJ2LS ha scritto:
> Looks like the frequency isn't fetched correctly. I did some changes to rigctld today. Are you 
> running latest code?

I was running the code updated until this morning

Now I have updated to this evening and frequency is shown

Franco Spinelli
IW2DHW

DJ2LS, 2024-10-01:
Nice! I‘m going to publish a release then I think. If everything’s fine then please close the issue. For explorer I have a separate issue open.Von meinem iPhone gesendet

I was running the code updated until this morning

Now I have updated to this evening and frequency is shown

Franco Spinelli
IW2DHW

closed 2024-10-02 by frspin