DJ2LS/FreeDATA #808: [Bug]: Frequency won't remain fixed when using VOX

danieldamon, 2024-09-16:
### FreeDATA Version

0.16.3-alpha

### What operating system are you using?

Other (specify below)

### Operating System Version

Linux and Windows

### Architecture

x64

### Area of problem

Server - General

### Describe the bug

When using VOX mode, the frequency should stay fixed after it is set. On Ubuntu version 0.16.3-alpha it stays momentarily then goes to NaN.  On Windows version 0.16.2-alpha, the same behavior can be seen

### To Reproduce

1. Set Rig control to VOX
2. use Frequency widget to set frequency at whatever is your choice.
3. Close widget and observe the frequency change back to unknown after a few seconds or minutes.

### Expected behavior

In order to report frequency on the FreeDATA explorer, the frequency must be known. So, in VOX mode, after being set by the frequency widget, it should remain fixed until changed.

DJ2LS, 2024-09-17:
Thanks for the feedback! 

Question is, could this be misleading, means, if a operator changes its frequency on the radio. So do we need an additional information somewhere which information,smithsonite the frequency is 8m this case just of informational character?

danieldamon, 2024-09-17:
smithonite?  Anyway, I'm pretty sure this is what other programs such as wstjx and varac to. For example, currently I'm in the middle of Arizona.  I'm having trouble with CAT, (not your problem), so I switched to VOX. But how can I indicate the frequency I'm running at if not manually inserted?

danieldamon, 2024-09-17:
(.. other programs do)

danieldamon, 2024-09-17:
Furthermore, the main reason I'm using VOX at the minute is because I don't have the USB to the transmitter isolated well enough.  In VOX, I'd expect communication to the transceiver to stop. However, I must unplug the USB to stop the noise from Freedata communication over the USB.

DJ2LS, 2024-09-17:
smithonite ... well, seems auto correction attacked me. I even don't know the meaning of this word. 

Okay, then it seems, it's making sense implementing this. Thanks for the feedback! 

What's the cause for the CAT problem? Any chance we can help?

DJ2LS, 2024-09-17:
Oh, I understand. I had this problem as well. However, let’s try to adjust the frequency thing.Thanks to your usb problems, the need for this adjustment is shown :-) Von meinem iPhone gesendet

danieldamon, 2024-09-17:
I deleted the comment about noise over the USB when in VOX mode. My mistake - as I was in standard Rigcontrol mode. I think I need to put ferrite beads on my USB line to stop the noise. Like I said, I don't think it is a Freedata isssue.

DJ2LS, 2024-09-18:
pushed a commit, can be tested in develop branch, otherwise it will be part of the next release ( 0.16.4 )

danieldamon, 2024-09-19:
Confirmed that it works in the development branch for Linux

closed 2024-09-21 by DJ2LS