DJ2LS/FreeDATA #809: [Feature Request]: Show multiple station in same QTH location on explorer map.

cris1, 2024-09-17:
### Problem Description

When two stations are in the same grid locator, (Maidenhead Locator System)  only 1 of the 2 stations becomes visible on the map.

### Proposed Solution

There are a number of possible solutions.

- Using an 8 number gridsquare maidenhead locator so that more precise location can be displayed on the map. This will show multiple pins on the map.

### Alternatives Considered

Other possibilities for achieving the same goal would be

- Changing the window display and making all stations in this QTH locator selectable. But still show 1 pin on the map.
- Specify GPS coordinates and use them for display on the map. (within the parameters of the gridsquare)
- Use random pin location within the “Grid square” per station.

### Additional Information

There are of course privacy considerations that must be kept in mind where there must always be an option not to choose a precise location determination . I think this should be a choice to be made by the end user.

DJ2LS, 2024-09-17:
Thanks for the detailed feature request! 

Well, a 8 character grid isn't possible as the freedata protocol supports only 6 characters. 

But there are plugins available for displaying this correctly on the map. 

Using GPS coordinates of the radio, like a IC705 for example, might be possible, but for now I think we should keep things simple.

DJ2LS, 2024-09-17:
https://github.com/jawj/OverlappingMarkerSpiderfier-Leaflet

This one is able handling with several pins at one position