DJ2LS/FreeDATA #810: [Feature Request]: FreeData server running under systemcontrol

pa1er, 2024-09-18:
### Problem Description

You have to run the server from a terminal. When using Linux connected to the TRX you always have to have an open terminal to keep it running. When starting the server under the systemctl  (freedata.service) this problem can be resolved in a simple way.

### Proposed Solution

Creating a freedata.service file and copy it to the right location to be able to start the server from systemctl

### Alternatives Considered

Any other way to let the server running out of a terminal

DJ2LS, 2024-09-18:
@pa1er thats an interesting feature request. Thanks!

@deejdeej what do you think regarding to this?

Personally I had several problems with services in past, so I'm concerned a bit about stability. Also the problems with debugging, specially for inexperienced users. Is there a chance, making this optional? Maybe a separated step part of the installation instructions? Nevertheless I like this as it's making the usage probably more lightweight as it is running in background ( this is a pro and con as well )

pa1er, 2024-09-18:
If you define the services than you don have to switch them on by default. In stead of starting with the run script you will than start with 'systemctl start freedata' or stop etc.

DJ2LS, 2024-09-18:
So to me it looks like, we just have to adjust the installation instructions as this is something each user can or should do by him/herself?

deejdeej, 2024-09-18:
You don't actually have to run the server from a terminal.  You can create a shortcut on the desktop and double click the icon to run it (which will then open a terminal automatically and run the software).

I could create a sample shortcut under the Linux tools section if you think that would be useful.  Here are manual instructions:

https://linuxconfig.org/how-to-create-desktop-shortcuts-in-linux

DJ2LS, 2024-09-18:
> If you define the services than you don have to switch them on by default. In stead of starting with the run script you will than start with 'systemctl start freedata' or stop etc. 
> 
> 

I think the service config should lead to the startup script so we have an equal way of starting FreeDATA

DJ2LS, 2024-09-18:
> You don't actually have to run the server from a terminal.  You can create a shortcut on the desktop and double click the icon to run it (which will then open a terminal automatically and run the [...]

Sounds to me like another options.

So we have two additional startup topics:
- as system service 
- by desktop icon

deejdeej, 2024-09-18:
I've added two files to the tools/Linux section in the develop branch.
README-Desktop-icon.txt - Instructions for setting up the Desktop icon
FreeDATA.desktop - the Desktop icon file

@DJ2LS Do we have a FreeDATA icon (graphics)?  If so, I can add that so it displays on the Desktop instead of the default symbol.

pa1er, 2024-09-18:
For me the desktop-item is not really working. I am using the Ubuntu server as a real server, so try to avoid using the screen. I connect with my laptop to a remote desktop OR the web interface of freeData. That's why systemservices are so great. And by by using the enable command (systemctl enable .....) you create the link for the startup after a reboot (or disable to remove the link)
By using a terminal or desktop-icon you always need the screen or a connection to a terminal. Systemservices keep running after connection to terminal or screen are gone. So  also nice to monitor the freq when you are away.

DJ2LS, 2024-09-18:
> I've added two files to the tools/Linux section in the develop branch.
>
> README-Desktop-icon.txt - Instructions for setting up the Desktop icon
>
> FreeDATA.desktop - the Desktop icon file
>
>
>
> @DJ2LS Do [...]

Yes, in docs folder or gui public folder

DJ2LS, 2024-09-18:
> For me the desktop-item is not really working. I am using the Ubuntu server as a real server, so try to avoid using the screen. I connect with my laptop to a remote desktop OR the web interface of [...]

Ah now I understand your use case. What if we create a custom service file in Linux/tools as well, then adding a command for symlink it to services folder? 

Then we have a example as well. But I think this is all just a question of documentation, right?

deejdeej, 2024-09-20:
Hi @pa1er ,
If you have a suggested freedata.service file, I can help you to test it and add it to the Linux tools section.  

Part of the challenge is that the existing run-freedata-linux.sh script expects to run in an interactive terminal, and waits for CTRL-C to exit.  I am not sure how to make that work in a systemd service file since those are designed to run non-interactive programs.

pa1er, 2024-09-20:
I will create a service that will work in my ubuntu server, For other things on my fedora server I have done this before. 
I'll let you know.

dk5sm, 2024-12-19:
> I will create a service that will work in my ubuntu server, For other things on my fedora server I have done this before. I'll let you know.

Have you had succes by now?

DJ2LS, 2025-02-27:
any news regarding this issue? 

We could add some instructions to the wiki I think. Doing this per default might cause some issues because of different distributions out there.

DJ2LS, 2025-05-06:
https://wiki.freedata.app/en/knowledge-base/raspberry-pi/install-as-system-service

Started working on this as part of my QMX+ with included raspberry pi 

@LA3QMA feel free adjusting it if you want. This might be of interest for you as well

closed 2025-05-24 by DJ2LS