DJ2LS/FreeDATA #811: [Bug]: partner station only responds on CQ

dk5sm, 2024-09-20:
### FreeDATA Version

0.16.4-alpha

### What operating system are you using?

Ubuntu

### Operating System Version

24.04.1

### Architecture

ARM Raspberry Pi

### Area of problem

Server - Message System

### Describe the bug

Yesterday I noticed, that I seem to have a good path to a Norwegian station; it´s beacon showed an SNR of +3dB. So i tried bothe  ping and chat, but both without response from IRS. Only a CQ led to a response from this particular station.

### To Reproduce

not sure how to do this, as ping/connect formerly worked the same day.

### Expected behavior

ping and chat should result in an corresponding answer ;-)

LA3QMA, 2024-09-20:
I can confirm this behavior. 19.09.2024 most of the stations i received was quite strong. After a CQ several stations reported and was updated in the list. Trying to send a message failed also ping. Can try to get the logs later todaty,

DJ2LS, 2024-09-20:
Are we talking about repeating or sending a new message?

LA3QMA, 2024-09-20:
I tried repeat and also a new message. Can try more later today to see if i can narrow it down.

LA3QMA, 2024-09-20:
hmm only CQ is working
and regarding messages in queued
2024-09-20 19:36:36 [info     ] [DatabaseManagerMessages]: Set message LA3QMA-1_DJ2LS-1_2024-09-18T17:37:35.168180+00:00 to queued and incremented attempt
2024-09-20 19:36:36 [warning  ] [DatabaseManagerMessages]: An error occurred while setting messages to queued: object of type 'P2PMessage' has no len()

DJ2LS, 2024-09-21:
Maybe this sticks together with the issue. If a message fails, it might have a wrong state. This results then into ignoring packages as we expect specific frames for specific states

DJ2LS, 2024-09-21:
> hmm only CQ is working and regarding messages in queued 2024-09-20 19:36:36 [info ] [DatabaseManagerMessages]: Set message LA3QMA-1_DJ2LS-1_2024-09-18T17:37:35.168180+00:00 to queued and incremented [...]

This one should be fixed now

closed 2024-09-25 by DJ2LS