DJ2LS/FreeDATA #812: [Bug]: chat or file transfer can be deleted during transfer

dk5sm, 2024-09-20:
### FreeDATA Version

0.16.4-alpha

### What operating system are you using?

Ubuntu

### Operating System Version

24.04.1

### Architecture

ARM Raspberry Pi

### Area of problem

Server - Message System

### Describe the bug

While a file transfer has already started, you can delete the file to be transferred from the chat. The transfer continues in the meantime.

### To Reproduce

1. go to (existing) chat
2. add a file + start transfer
3. hit the wastebin-icon to delete the file

### Expected behavior

It is contradictory to be able to delete a file/chat while it is still being transferred.

I´d suggest a kind of file 'locking': hide the delete button while the transfer is running OR hitting the delete button cancels the transfer and deletes the file.

DJ2LS, 2024-09-20:
Suggestions, @dk5sm @LA3QMA @deejdeej @Mashintime ?

I tend to "stop" and delete.

DJ2LS, 2025-03-18:
Part of latest code in development

closed 2025-04-05 by DJ2LS