DJ2LS/FreeDATA #819: [Bug]:

OH2JRB, 2024-09-28:
### FreeDATA Version

0.16.4-alpha

### What operating system are you using?

Other Linux

### Operating System Version

Debian Bookworm

### Architecture

x86

### Area of problem

Server - Message System

### Describe the bug

Log flooded - repeating message several times in every 10 seconds. Does not know whether this state setting is needed this frequently.

### To Reproduce

Encountered in normal operations.

### Expected behavior

Clean log ...

### Screenshots

2024-09-28 06:06:33 [warning  ] [SCHEDULE] [ARQ=85] Setting state to old_state=<IRS_State.RESUME: 7> state=<IRS_State.RESUME: 7>
2024-09-28 06:06:33 [info     ] [ARQSessionIRS][id=85][state=IRS_State.RESUME]: ARQSessionIRS state RESUME unchanged.
2024-09-28 06:06:33 [warning  ] [SCHEDULE] [ARQ=121] Setting state to old_state=<IRS_State.RESUME: 7> state=<IRS_State.RESUME: 7>
2024-09-28 06:06:33 [info     ] [ARQSessionIRS][id=121][state=IRS_State.RESUME]: ARQSessionIRS state RESUME unchanged.
2024-09-28 06:06:33 [warning  ] [SCHEDULE] [ARQ=15] Setting state to old_state=<IRS_State.RESUME: 7> state=<IRS_State.RESUME: 7>
2024-09-28 06:06:33 [info     ] [ARQSessionIRS][id=15][state=IRS_State.RESUME]: ARQSessionIRS state RESUME unchanged.
2024-09-28 06:06:33 [warning  ] [SCHEDULE] [ARQ=109] Setting state to old_state=<IRS_State.RESUME: 7> state=<IRS_State.RESUME: 7>
2024-09-28 06:06:33 [info     ] [ARQSessionIRS][id=109][state=IRS_State.RESUME]: ARQSessionIRS state RESUME unchanged.

DJ2LS, 2024-09-28:
This issue occurs for me as well. Will try to work on this today

DJ2LS, 2024-09-28:
should be fixed now

closed 2024-09-28 by DJ2LS