DJ2LS/FreeDATA #827: [Bug]: No Frequency with external Hamlib

OH1KH, 2024-09-30:
### FreeDATA Version

? (from main branch 2024-09-30)

### What operating system are you using?

Other Linux

### Operating System Version

Linux hamtpad 6.10.11-200.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Sep 18 21:09:58 UTC 2024 x86_64 GNU/Linux

### Architecture

x64

### Area of problem

Rig Control

### Describe the bug

Usually start rigctld with script to be able to access rig for many programs (same time).
/usr/local/bin/rigctld -m 3073 -r /dev/icom7300 -t 4532 -s 19200 -C auto_power=0 --vfo -vvvvv -Z

If started this way FreeData/external hamlib (127.0.0.1/4532) shows that rig is online.
How ever no frequency is shown and can not be changed.
Using switch "vfo" does not give any change.

From rigctld debug I see that FreeData is accessing rigctld and tries to dig out ALC, Power, etc. with no response.
During testing it also sent CW via hamlib randomly "ANKS?1". It seem to be collision in commands sent to external rigctld.

Using "Hamlib internal" and IC7300 settings it seems to work ok. This is how ever what I do not want as I need rig access also with other programs (same time).

### To Reproduce

Start rigctld from script first.
Then start FreeData using "Hamlib external" (127.0.0.1/4532)

### Expected behavior

See frequency and to be able to set it from program

DJ2LS, 2024-09-30:
@LA3QMA @Mashintime can you help me with this? On my Mac rigctld seems not to work properly

OH1KH, 2024-09-30:
I debugged this a bit.

My FreeData version is:  API Version: 3 Modem Version: 0.16.6-alpha

There are some problems when "Hamlib external" is used.
First:
At init command "\chk_vfo" **must** be issued.
That gives information how remote rigctld has been started.
And with that information following commands must be formatted.
if \chk_vfo is "0" (zero) we can ask frequency with "f" and set it wit "F"
if \chk_vfo is "1" we must add vfo information to request. 
The easiest way is to send command "f currVFO" to get frequency and "F currVFO" to set it.

using "currVFO" instead of true vfo name is safe as ICOM rigs (mostly all) can not tell what VFO is in use. (get_VFO is not supported command)

There is slider "vfo" in "rig vfo parameter" and it looks like user could define the same thing as "\chk_vfo" command can give.
How ever what the slider position is FreeData polls frequency always using only "f" and this does not work if external rigctld has been started with "--vfo" parameter.

**This must be fixed!**

Second:
When external connection is used we do not know what rig model there may be. 
That is why command \dump_caps should be issued first. It gives a long list of rig properties and from that we can look "Can get" to see is it worth of polling ALC, Power etc. issues.

While debug I did not see \dump_caps issued. 

Maybe this should be done?

DJ2LS, 2024-09-30:
Thanks for the support, @OH1KH 
Just saving some thoughts 
```
chk_vfo
Returns “CHKVFO 1\n” (single line only) if rigctld was invoked with the -o/--vfo option and “CHKVFO 0\n” if not.
When in VFO mode the client will need to pass 'VFO' as the first parameter to set or get commands. VFO is one of the strings defined in set_vfo abo
```

DJ2LS, 2024-09-30:
@OH1KH 

I think we should work on both of your suggestions.

I was missing the chk_vfo options, thanks for finding it! I think this part might be the easier one.

A bit more challenging might be the selective parameter polling using the dump command. But I guess it's worth doing it

DJ2LS, 2024-09-30:
well, `dump_caps` command doesn't seem to run that fine. From rigctl it's possible but via Python it's getting stuck for some reason.

DJ2LS, 2024-10-01:
chk_vfo is also not working reliably. Really a pain

DJ2LS, 2024-10-01:
okay, it seems, chk_vfo is working, but hamlib has a warmup phase where its returning wrong values

DJ2LS, 2024-10-01:
it seems the latest version is now running and correctly detecting VFO setting by chk_vfo.

The output of dump_caps seems to be working as well, but it's a minor step of work, so I'd like to wait with work on this. I just disabled the dump_caps code, so we might work on this later. But I suggest, let's open an additional issue for this, so we can work later. 

If this Issue is working for you, @OH1KH , we can close it

closed 2024-10-01 by OH1KH

DJ2LS, 2024-10-01:
> HI!
> "devel" branch version seems to work now using "Rigctld(external hamlib) "
> Using:  rigctld Hamlib 4.6~git 2024-09-29T03:50:25Z SHA=a6fe6b 64-bit
> Started as: /usr/local/bin/rigctld -m 3073 -r /dev/icom7300 -t 4532 -s 19200 -C auto_power=0 --vfo -vvvvv -Z
> Also "Rigctld(internal hamlib)" seems to work when proper values is set for IC7300 connected (as it was also before)
> So far, so good now.
> Thanks!
> P.S. still wondering the purpose of  "vfo" slider at Settings/rigctld VFO parameter. It seems to have no effect at all.

It lets you start rigctld with vfo option for supporting fldigi usage Von meinem iPhone gesendet