DJ2LS/FreeDATA #829: [Bug]: Internal hamlib reports OK when transmitter is disconnected

danieldamon, 2024-09-30:
### FreeDATA Version

0.16.8-alpha

### What operating system are you using?

Ubuntu

### Operating System Version

Ubuntu 24.04

### Architecture

x64

### Area of problem

Rig Control

### Describe the bug

Without the radio connected, FreeData reports everything is operational. I have rig set for TS480

### To Reproduce

1. disconnect radio
2. start FreeData
3. set rigctl to internal hamlib
4. See all green

### Expected behavior

Without a rig, indicator should be red

### Screenshots

none

danieldamon, 2024-09-30:
This issue does not appear in Windows version 0.16.2-alpha, which was downloaded today.

DJ2LS, 2024-10-01:
This issue sticks together with #827

DJ2LS, 2024-10-01:
latest development code should work now. Can you confirm this, @danieldamon ? Otherwise we have to wait until we published a new release.

danieldamon, 2024-10-01:
Sorry, continues the same incorrect way. 
I built by first deleting contents of FreeDATA, then executing
bash ./install-freedata-linux.sh develop

DJ2LS, 2024-10-01:
Question is, if it's a Hamlib related topic. Means, if Hamlib is running fine, but just can't connect to the radio, it's hard to avoid the behaviour as we are checking the network connection. Not sure if we have a Hamlib related api command

danieldamon, 2024-10-01:
Internal hamlib appears to be working correctly.  It changes frequency of my rig and causes it to transmit. At least on non-development.  I'll try there too

danieldamon, 2024-10-01:
same deal with development

closed 2024-11-11 by DJ2LS