DJ2LS/FreeDATA #835: Structured Bug hunting

DJ2LS, 2024-10-07:
This issue should help, checking all FreeDATA components in a structured way. If a component is going to be inspected, it can be marked as checked and new issues can be opened in case of bugs. There is no timeline, but work should focus on this continuously 

It's a work in progress yet!

### GUI General Layout

### GUI Main screen

### GUI Chat Screen

### GUI Settings Screen

### Server

HB9HBO, 2025-02-01:
I've do several tests regarding to the Linux installation.
Debian 11, install OK, works, CPU normal
Debian 12, install OK, works, CPU normal
Ubuntu 22, not tested
Ubuntu 24, install OK, works, CPU normal
Fedora 40, not tested
Fedora 41, works if VERSION_ID=41 is added to the Fedora install section.
Raspberry Pi 4 aarch64, install OK, works, but CPU is over 95% most of the time.

Two notes:
Maybe the script should check if the user is in the sudo group. This would help newbies.
Depending on the used transceiver, the user needs to be in the dialout group. If it is not, everything works fine, but the radio would not respond.
Or just mention this in the installation dokumentation.

deejdeej, 2025-02-01:
Added support for Fedora 41 to the install script.

deejdeej, 2025-02-01:
I've added a check for the account being in the dialout group in the install script, and exits with a warning if it isn't there.  Checking if an account has sudo privs is a little more complicated, so I opted to simply write out warning on the screen saying if the sudo command fails with "Sorry", or "not in sudoers file", or something to that effect, to make sure your account has sudo privs.  At least this gives a hint of where to look if there is an issue.

DJ2LS, 2025-03-04:
@LA3QMA any news regarding deleting messages? If its working, just let me know, so we ken mark this task as "checked"

LA3QMA, 2025-03-04:
The message is now deleted from the table "p2p_message" and "message_attachment" when i delete it from a individual message. But the attachments is still in the table "attachment". Using the "Delete Chat" i think also deleting the attachments would be useful

DJ2LS, 2025-03-04:
Also at restart there is a check now if orphaned attachments are there. Maybe this solves this in your case because of old unclean state

closed 2025-04-16 by DJ2LS